RT-Thread/rt-thread #3406: 原子407开发板的U盘功能, 有人成功过吗?

ylz0923, 2020-02-25:
拉取最新带驱动, 我的原子407开发板的U盘功能时无效的, 不知是驱动问题还是我的开发板有问题?

QQWE1345, 2020-03-08:
Did you succeed?  I have the same problem。
I successfully port it on the basis of lts 3.1.4, and the test pass.
Now I consider what's wrong with the V4.0.3

BernardXiong, 2020-03-14:
Is it failed on v4.0.3?

thread-liu, 2020-03-14:
使用 USB HOST 功能时，请确保 USB_PWR 引脚为高电平。否则无法识别到外部设备。407 开发板的 USB_PWR 脚为 PA15 。初始化 PA15, 并让其输出高电平。就可以正常使用 U 盘功能啦。

```
    GPIO_InitStruct.Pin = GPIO_PIN_15;
    GPIO_InitStruct.Mode = GPIO_MODE_OUTPUT_PP;
    GPIO_InitStruct.Pull = GPIO_NOPULL;
    HAL_GPIO_Init(GPIOA, &GPIO_InitStruct);
    
    HAL_GPIO_WritePin(GPIOA,0x01<<15UL,GPIO_PIN_SET);
```

终端打印：

```

 \ | /
- RT -     Thread Operating System
 / | \     4.0.3 build Mar 14 2020
 2006 - 2020 Copyright by rt-thread team
msh />Mount FAT on Udisk successful.

msh />ls
Directory /:
System Volume Information<DIR>                    
HiTrendtech_M3_BSP_and_Note_2020-2-24.rar16434252                 
.ZIP        5873056                  
msh />
```

SummerLife, 2020-03-16:
@ylz0923 通过 thread-liu 的方法问题有解决吗？

ylz0923, 2020-03-16:
USB_PWR 我是设置好了的, 你们的板子可以用, 应该是我的开发板坏了

closed 2020-03-16 by ylz0923

QQWE1345, 2020-03-16:
> Is it failed on v4.0.3?

no it work very well. I have solved the problem.
Here is my post .
https://www.rt-thread.org/qa/thread-423804-1-1.html