RT-Thread/rt-thread #3409: file 线程是在哪建的，怎么调这个线程栈的大小？

SuWeipeng, 2020-02-26:
没找到 file 线程创建的地方，求助。
---
[image]

SuWeipeng, 2020-02-26:
我有用文件系统在 SD 卡上记日志，但我没有建过叫“file”的线程。
---
[image]

shao7936626, 2020-02-26:
什么叫file的线程？   你哪个线程去用file去读文件，就是file线程啊，那个线程就要弄大一点，不然读的文件大了，就溢出了

SuWeipeng, 2020-02-26:
> 什么叫file的线程？ 你哪个线程去用file去读文件，就是file线程啊，那个线程就要弄大一点，不然读的文件大了，就溢出了

怎么调这个线程的大小呢？我没找到创建线程的地方，所以不知道在哪调。

SuWeipeng, 2020-02-26:
> > 什么叫file的线程？ 你哪个线程去用file去读文件，就是file线程啊，那个线程就要弄大一点，不然读的文件大了，就溢出了
> 
> 怎么调这个线程的大小呢？我没找到创建线程的地方，所以不知道在哪调。

找到了，在我写文件的地方。

closed 2020-02-27 by SuWeipeng