RT-Thread/rt-thread #3429: consle烧写完或者软重启不能用

SmallSmartMouse, 2020-03-05:
使用的是4.0.4的系统，芯片是stm32f427，在设备烧写完成后，consle不能输入，通过复位按钮也是一样的情况，但是能够输出，
断电重启后能够输入，但是会丢失程序刚开始打印的一部分内容。

armink, 2020-03-05:
是不是原子的板子？注意下 boot 电路

SmallSmartMouse, 2020-03-05:
不是，在单步的情况下，测试能够进consle串口的中断，但是consle的信息没有处理，估计是内核问题，猜测内核有些初始化采用了多线程的初始化，但是有依赖的模块没有保证时序

SmallSmartMouse, 2020-03-05:
系统用的是4.0.2的系统

BernardXiong, 2020-03-06:
shell部分依赖关系不强，只要串口驱动OK，shell就可以使用

建议按照这样的顺序检查下：

1. v4.0.2是否有问题；
2. github/master版本是否有问题？

都有问题，单步step by step方式检查下可能是哪里的问题，排除掉本身环境的因素，如果有问题，可以修复了提交PR上来修复它。

SmallSmartMouse, 2020-03-06:
测试发现 master分支的代码也是一样的情况，
单步测试只能看到串口中断收到了数据，并且放到了buffer里面，但是不清楚shell的处理入口在哪里，不知道在哪里打断点

SmallSmartMouse, 2020-03-10:
这个开源系统的响应反馈这么消极吗？已经在各个渠道反馈系统的各种问题了，没有一个积极响应的，已经决定换系统了。

BernardXiong, 2020-03-14:
github相对慢的

shell那边是通过shell.c/finsh_getchar 函数来获得输入

SmallSmartMouse, 2020-03-14:
`static int finsh_getchar(void)
{
#ifdef RT_USING_DEVICE
#ifdef RT_USING_POSIX
    return getchar();
#else
    char ch = 0;

    RT_ASSERT(shell != RT_NULL);
    while (rt_device_read(shell->device, -1, &ch, 1) != 1)
        rt_sem_take(&shell->rx_sem, RT_WAITING_FOREVER);

    return (int)ch;
#endif
#else
    extern char rt_hw_console_getchar(void);
    return rt_hw_console_getchar();
#endif
}`
重点是不知道getchar()的实现是在哪里，测试发现能够进串口接收中断，但是不能进这个位置的断点

BernardXiong, 2020-03-14:
可以暂时把RT_USING_POSIX的宏拿掉，然后进入到后面的rt_device_read。

因为你是stm32f427的芯片，这个需要明确你的BSP是怎么来的，移植是否对，特别是和POSIX相关的移植是否一致，是否使用scons & menuconfig模式来整工程的。

SmallSmartMouse, 2020-03-14:
`            rt_wqueue_wait(&(device->wait_queue), 0, RT_WAITING_FOREVER);
`
经过测试，代码收到数据后一直在等待队列

SmallSmartMouse, 2020-03-14:
测试发现，插着sd卡重启不能试用consle，拔掉重启就可以用

BernardXiong, 2020-03-14:
这个就不得而知了，你提供的信息太少了，能给到的建议也同样的少

SmallSmartMouse, 2020-03-14:
`        rt_device_open(new_device, RT_DEVICE_OFLAG_RDWR | RT_DEVICE_FLAG_STREAM | RT_DEVICE_FLAG_INT_RX);
`
添加了一个 RT_DEVICE_FLAG_INT_RX参数，consle可以输入了，但是断电重启的时候刚启动的一段打印信息还是没有输出，过一会才会打印循环输出的信息，另外还是不理解为什么sd卡的代码会影响consle，系统各个模块之间耦合太多了。

SmallSmartMouse, 2020-03-14:
`rt_inline int _serial_int_tx(struct rt_serial_device *serial, const rt_uint8_t *data, int length)
{
    int size;
    struct rt_serial_tx_fifo *tx;

    RT_ASSERT(serial != RT_NULL);

    size = length;
    tx = (struct rt_serial_tx_fifo*) serial->serial_tx;
    RT_ASSERT(tx != RT_NULL);

    while (length)
    {
        if (serial->ops->putc(serial, *(char*)data) == -1)
        {
            rt_completion_wait(&(tx->completion), RT_WAITING_FOREVER);
            continue;
        }

        data ++; length --;
    }

    return size - length;
}`

`rt_inline int _serial_poll_tx(struct rt_serial_device *serial, const rt_uint8_t *data, int length)
{
    int size;
    RT_ASSERT(serial != RT_NULL);

    size = length;
    while (length)
    {
        /*
         * to be polite with serial console add a line feed
         * to the carriage return character
         */
        if (*data == '\n' && (serial->parent.open_flag & RT_DEVICE_FLAG_STREAM))
        {
            serial->ops->putc(serial, '\r');
        }

        serial->ops->putc(serial, *data);

        ++ data;
        -- length;
    }

    return size - length;
}
`
`    if (dev->open_flag & RT_DEVICE_FLAG_INT_RX)
    {
        return _serial_int_rx(serial, (rt_uint8_t *)buffer, size);
    }
#ifdef RT_SERIAL_USING_DMA
    else if (dev->open_flag & RT_DEVICE_FLAG_DMA_RX)
    {
        return _serial_dma_rx(serial, (rt_uint8_t *)buffer, size);
    }
#endif /* RT_SERIAL_USING_DMA */    

    return _serial_poll_rx(serial, (rt_uint8_t *)buffer, size);`

启用 RT_DEVICE_FLAG_INT_RX标识的情况下，和没有启用的情况下，代码不一致啊，一个碰到\n会补充\r，另一个不会，这个代码的问题实在是太多了

BernardXiong, 2020-03-15:
请先仔细看文档，以及我上面提及到的部分。没有相关的一些信息，这些问题都是无从谈起的

closed 2020-03-15 by BernardXiong