RT-Thread/rt-thread #3459: 用户写中断服务程序有多少栈空间可用呢？

SuWeipeng, 2020-03-16:
[在这里](https://www.rt-thread.org/document/site/programming-manual/interrupt/interrupt/)看到了这段话：
[image]

# “中断栈”的大小能在哪里知道呢

SuWeipeng, 2020-03-16:
[image]

SuWeipeng, 2020-03-16:
这样理解是否正确呢？
---
[image]

SuWeipeng, 2020-03-16:
问题的由来
---
使用 timer，超时回调函数里想写数组，有循环。ms 级定时，这种情况用 soft 模式定时器更好。

当看到 soft 模式在 env 里有配置线程栈大小和优先级，就联想到 hard 模式应该也能配置栈大小。但没有找到有关**中断栈**设置的相关信息。所以就来这里提问了。

BernardXiong, 2020-03-17:
中断栈是和ISR程序相关的，一般是由工程配置来决定，例如一些Cortex-M的BSP，是由startup.S的启动汇编/链接脚本来指定的。

* Keil MDK, for example Stack_Size in startup_stm32f103xe.S；
* gcc, for example _system_stack_size in link.lds.

目前没有配置项，以及也还未有命令来查看中断栈已经用到多少了。

closed 2021-02-24 by mysterywolf