RT-Thread/rt-thread #3471: IAR專案有檔案名稱相同但不同路徑重名問題?

wosayttn, 2020-03-20:
請問這個要何如解決?
components\drivers\usb\usbdevice\core\core.c
components\drivers\usb\usbhost\core\core.c

Building configuration:  - Release
Updating build tree...
**Build error: **Multiple tools write to the same file.****
  File: C:\Program Files\Jenkins\jobs\RTThread IAR Weekly Build
  Tool 1: C/C++ Compiler, when processing C:\Program Files\Jenkins\jobs\RTThread IAR Weekly Build\workspace\components\drivers\usb\usbdevice\core\core.c
  Tool 2: C/C++ Compiler, when processing C:\Program Files\Jenkins\jobs\RTThread IAR Weekly Build\workspace\components\drivers\usb\usbhost\core\core.c
0	file(s) deleted.

BernardXiong, 2020-03-22:
这个无解，IAR在这个方面支持奇差

closed 2020-03-29 by BernardXiong

wosayttn, 2020-04-28:
是否變更檔名為： ud_core.c 和 uh_core.c ?    若ok，我搞個PR。

mysterywolf, 2022-02-17:
Keil IAR都不允许同名.c存在