RT-Thread/rt-thread #3482: interface 0 limitation in hid.c

egbert-h, 2020-03-26:
hi,

Does anyone know why the limitation that interface should be 0 in the line 461,462 of the hid.c?

`    if(setup->wIndex != 0)
        return -RT_EIO;`

I tested a composite device with classes(CDC+HID)
and removed the "if expression" and the composite device worked well.

If this expression is in code, the host cannot get hid report descriptor.

qz721, 2020-03-26:
Thanks for your remind.

This is a bug. The driver engineer might have misunderstand the `Get_Descriptor` request and the `Get_Report` one. The `if` statement should be removed.

It will be fixed before long. Thank you again!

Guozhanxin, 2020-12-23:
@qz721 Have you fixed this problem?

balanceTWK, 2020-12-28:
https://github.com/RT-Thread/rt-thread/commit/a49ddaef96a26edc0c1afcc5e260effcf709cd31#diff-8867ba0122a0eb11dac95a3987826712723835a9fd02fde1f1ff74fdbd3eff09

The problem has been fixed, you can open it again if you have any questions!

closed 2020-12-28 by balanceTWK