RT-Thread/rt-thread #3483: Compiling for RV64(K210, KD233)， met a error: unknown type name 'siginfo_t'

tekkamanninja, 2020-03-26:
CC build/kernel/components/drivers/misc/pin.o
In file included from /home/tekkamanninja/development/RISC-V/K210/github/rt-thread/include/rtthread.h:24,
                 from /home/tekkamanninja/development/RISC-V/K210/github/rt-thread/components/drivers/include/drivers/pin.h:15,
                 from /home/tekkamanninja/development/RISC-V/K210/github/rt-thread/components/drivers/misc/pin.c:11:
/home/tekkamanninja/development/RISC-V/K210/github/rt-thread/include/rtdef.h:471:9: error: unknown type name 'siginfo_t'
 typedef siginfo_t rt_siginfo_t;
         ^~~~~~~~~
scons: *** [build/kernel/components/drivers/misc/pin.o] Error 1
scons: building terminated because of errors.

qz721, 2020-03-26:
The structure `siginfo_t` itself is defined both in the kernel code and the kernel library, one of them will be chosen to be compiled depending on your configuration.

This problem is usually caused by an inappropriate version of compiler, please tell me the version you're using and show me the content of `cconfig.h`.

closed 2020-03-26 by tekkamanninja

tekkamanninja, 2020-03-26:
> The structure `siginfo_t` itself is defined both in the kernel code and the kernel library, one of them will be chosen to be compiled depending on your configuration.
>
> This problem is usually caused [...]

Hi thanks for your reply, I have tried it out, 

on the master branch, the problem is between 3d5357bc7 and 720d8479a, I mean :
I can successful compile the RTT by "git checkout 57ff16edb"
But once I "git checkout 4cffe19df"  , the error occurs. 

3d5357bc7 [AT] Fix typos in Kconfig
4cffe19df Merge pull request #3364 from BernardXiong/gcc_version
======================================================
57ff16edb Merge pull request #3365 from longtengmcu/master
720d8479a 给软件定时器增加定时器工作状态查询的功能

I am using gnu-mcu-eclipse-riscv-none-gcc-8.2.0-2.2-20190521-0004-centos64 on Fedora 31.

qz721, 2020-03-26:
OK, I'll report this issue.

And I've been told that the supports for GCC 8.x and 9.x are far from perfect. Could you try other versions of GCC and tell me whether the same issue comes up ?

Thank you for your efforts.

tekkamanninja, 2020-03-26:
> OK, I'll report this issue.
> 
> And I've been told that the supports for GCC 8.x and 9.x are far from perfect. Could you try other versions of GCC and tell me whether the same issue comes up ?
> 
> Thank you for your efforts.

more info :
once I revert this patch 
https://github.com/RT-Thread/rt-thread/pull/3364/commits/835504ed956bd6528cbd473ae2fe4b0f650c7188

the master branch can be compiled successfully, I mean 
MAYBE the fix is this :

diff --git a/tools/gcc.py b/tools/gcc.py
index 8c9685d52..71921bee2 100644
--- a/tools/gcc.py
+++ b/tools/gcc.py
@@ -162,7 +162,7 @@ def GCCResult(rtconfig, str):
             result += '#define HAVE_SIGVAL 1\n'
 
         if version:
-            result += '#define GCC_VERSION_STR "%s"\n' % version
+            result += '#define GCC_VERSION "%s"\n' % version
 
         result += '#define STDC "%s"\n' % stdc

tekkamanninja, 2020-03-26:
6cbbc6cdc75f84584ff701f10a550787038cdedc

qz721, 2020-03-26:
I'll testing serval versions of GCC and will paste the result here.

Please reopen the issue until we solve it.

qz721, 2020-03-27:
I have test the relative commits, on ARMv7 platform with the BSP for `qemu-vexpress-a9`, the result is here:
```
## Test commit 3d5357bc755c23ee5ebdf6b5471d8cacf894d506 ##
GCC version: gcc-arm-none-eabi-4_6-2012q4...          SUCCEEDED!
GCC version: gcc-arm-none-eabi-4_7-2012q4...          SUCCEEDED!
GCC version: gcc-arm-none-eabi-4_8-2014q3...          SUCCEEDED!
GCC version: gcc-arm-none-eabi-5_4-2016q3...          SUCCEEDED!
GCC version: gcc-arm-none-eabi-6-2017-q2-update...    SUCCEEDED!
GCC version: gcc-arm-none-eabi-7-2018-q2-update...    FAILED!
GCC version: gcc-arm-none-eabi-8-2019-q3-update...    FAILED!

## Test commit 4cffe19dfe04f30fff2bf623e840c44f45ed418c ##
GCC version: gcc-arm-none-eabi-4_6-2012q4...          SUCCEEDED!
GCC version: gcc-arm-none-eabi-4_7-2012q4...          SUCCEEDED!
GCC version: gcc-arm-none-eabi-4_8-2014q3...          SUCCEEDED!
GCC version: gcc-arm-none-eabi-5_4-2016q3...          SUCCEEDED!
GCC version: gcc-arm-none-eabi-6-2017-q2-update...    SUCCEEDED!
GCC version: gcc-arm-none-eabi-7-2018-q2-update...    FAILED!
GCC version: gcc-arm-none-eabi-8-2019-q3-update...    FAILED!

## Test commit 57ff16edb926948c647e3ec7b6f9eccf186cf98f ##
GCC version: gcc-arm-none-eabi-4_6-2012q4...          SUCCEEDED!
GCC version: gcc-arm-none-eabi-4_7-2012q4...          SUCCEEDED!
GCC version: gcc-arm-none-eabi-4_8-2014q3...          SUCCEEDED!
GCC version: gcc-arm-none-eabi-5_4-2016q3...          SUCCEEDED!
GCC version: gcc-arm-none-eabi-6-2017-q2-update...    SUCCEEDED!
GCC version: gcc-arm-none-eabi-7-2018-q2-update...    FAILED!
GCC version: gcc-arm-none-eabi-8-2019-q3-update...    FAILED!

## Test commit 720d8479a574746020747ef36bc33a8159206289 ##
GCC version: gcc-arm-none-eabi-4_6-2012q4...          SUCCEEDED!
GCC version: gcc-arm-none-eabi-4_7-2012q4...          SUCCEEDED!
GCC version: gcc-arm-none-eabi-4_8-2014q3...          SUCCEEDED!
GCC version: gcc-arm-none-eabi-5_4-2016q3...          SUCCEEDED!
GCC version: gcc-arm-none-eabi-6-2017-q2-update...    SUCCEEDED!
GCC version: gcc-arm-none-eabi-7-2018-q2-update...    FAILED!
GCC version: gcc-arm-none-eabi-8-2019-q3-update...    FAILED!

## Test commit e7af915cf2e05fd4fc38be4636a9637f98d37072 ##
GCC version: gcc-arm-none-eabi-4_6-2012q4...          SUCCEEDED!
GCC version: gcc-arm-none-eabi-4_7-2012q4...          SUCCEEDED!
GCC version: gcc-arm-none-eabi-4_8-2014q3...          SUCCEEDED!
GCC version: gcc-arm-none-eabi-5_4-2016q3...          SUCCEEDED!
GCC version: gcc-arm-none-eabi-6-2017-q2-update...    SUCCEEDED!
GCC version: gcc-arm-none-eabi-7-2018-q2-update...    FAILED!
GCC version: gcc-arm-none-eabi-8-2019-q3-update...    FAILED!

## Test commit 835504ed956bd6528cbd473ae2fe4b0f650c7188 ##
GCC version: gcc-arm-none-eabi-4_6-2012q4...          SUCCEEDED!
GCC version: gcc-arm-none-eabi-4_7-2012q4...          SUCCEEDED!
GCC version: gcc-arm-none-eabi-4_8-2014q3...          SUCCEEDED!
GCC version: gcc-arm-none-eabi-5_4-2016q3...          SUCCEEDED!
GCC version: gcc-arm-none-eabi-6-2017-q2-update...    SUCCEEDED!
GCC version: gcc-arm-none-eabi-7-2018-q2-update...    FAILED!
GCC version: gcc-arm-none-eabi-8-2019-q3-update...    FAILED!

## Test commit a7c590773e1be14e163ca67260c8927024c985dd ##
GCC version: gcc-arm-none-eabi-4_6-2012q4...          SUCCEEDED!
GCC version: gcc-arm-none-eabi-4_7-2012q4...          SUCCEEDED!
GCC version: gcc-arm-none-eabi-4_8-2014q3...          SUCCEEDED!
GCC version: gcc-arm-none-eabi-5_4-2016q3...          SUCCEEDED!
GCC version: gcc-arm-none-eabi-6-2017-q2-update...    SUCCEEDED!
GCC version: gcc-arm-none-eabi-7-2018-q2-update...    FAILED!
GCC version: gcc-arm-none-eabi-8-2019-q3-update...    FAILED!

## Test commit cd8dd40cd2dc0adf25ca684071ce082265de4cf1 ##
GCC version: gcc-arm-none-eabi-4_6-2012q4...          SUCCEEDED!
GCC version: gcc-arm-none-eabi-4_7-2012q4...          SUCCEEDED!
GCC version: gcc-arm-none-eabi-4_8-2014q3...          SUCCEEDED!
GCC version: gcc-arm-none-eabi-5_4-2016q3...          SUCCEEDED!
GCC version: gcc-arm-none-eabi-6-2017-q2-update...    SUCCEEDED!
GCC version: gcc-arm-none-eabi-7-2018-q2-update...    FAILED!
GCC version: gcc-arm-none-eabi-8-2019-q3-update...    FAILED!

## Test commit e5aafd2aa0b6165693f51be3d3c5b6f699558c69 ##
GCC version: gcc-arm-none-eabi-4_6-2012q4...          SUCCEEDED!
GCC version: gcc-arm-none-eabi-4_7-2012q4...          SUCCEEDED!
GCC version: gcc-arm-none-eabi-4_8-2014q3...          SUCCEEDED!
GCC version: gcc-arm-none-eabi-5_4-2016q3...          SUCCEEDED!
GCC version: gcc-arm-none-eabi-6-2017-q2-update...    SUCCEEDED!
GCC version: gcc-arm-none-eabi-7-2018-q2-update...    FAILED!
GCC version: gcc-arm-none-eabi-8-2019-q3-update...    FAILED!
```

It seems only the GCC version matters on ARM. Could you try your BSP with a lower version of GCC?

qz721, 2020-03-27:
Here is the script I used to test GCCs:
```shell
#!/usr/bin/bash
#set -e

gccdirs="$(find ~/rttwork/compilers/ -maxdepth 2 -type d -name *-20* -print | sort)"

function try_gcc_versions {
for d in $(echo -n $gccdirs | sort); do
  printf %-54s "GCC version: ${d##*/}... "
  export RTT_EXEC_PATH="$d/bin/"
  pushd ~/rttwork/rt-thread/bsp/qemu-vexpress-a9 >/dev/null
  scons -c >/dev/null
  git checkout $1 >/dev/null
  scons >/dev/null && echo SUCCEEDED! || echo FAILED!
    #( echo FAILED!; scons -c >/dev/null; scons 2>&1 >/dev/null; )>&5
  popd >/dev/null
done
}

commits="3d5357bc755c23ee5ebdf6b5471d8cacf894d506 \
  4cffe19dfe04f30fff2bf623e840c44f45ed418c \
  57ff16edb926948c647e3ec7b6f9eccf186cf98f \
  720d8479a574746020747ef36bc33a8159206289 \
  e7af915cf2e05fd4fc38be4636a9637f98d37072 \
  835504ed956bd6528cbd473ae2fe4b0f650c7188 \
  a7c590773e1be14e163ca67260c8927024c985dd \
  cd8dd40cd2dc0adf25ca684071ce082265de4cf1 \
  e5aafd2aa0b6165693f51be3d3c5b6f699558c69"

for c in $commits; do
  echo "## Test commit $c ##"
  try_gcc_versions $c
  echo
done >test.log
```