RT-Thread/rt-thread #3485: 'NoneType' object has no attribute 'groups' when switching python2 to python3

imxood, 2020-03-26:
if I use python3,  scons --menuconfig, or scons --target=eclipse, all is wrong.
[image]

however, using python2, it is OK.
[image]

SummerLife, 2020-03-26:
也许你手动升级一下 env 的脚本就没这样的问题了

imxood, 2020-03-27:
> 也许你手动升级一下 env 的脚本就没这样的问题了

imxood, 2020-03-27:
> 也许你手动升级一下 env 的脚本就没这样的问题了

我是ubuntu系统
你说的是这样的吗:
```
source ~/.env/env.sh
pkgs --update
```
又试了一遍, 也是不行, 一样的结果.

SummerLife, 2020-03-27:
不是这个意思，我的意思是 pkgs --upgrade 更新 env 功能脚本到最新版，

或者你切换到 env 的功能脚本目录，使用 git pull 命令来更更新

imxood, 2020-04-15:
现在可以了, 或许 apt install scons 默认安装的是python2版本的, 需要卸载scons, apt autoremove scons, 安装py3的, pip3 install scons, 就可以使用scons编译了

closed 2020-04-15 by imxood

mkx171217, 2022-10-21:
> 现在可以了, 或许 apt install scons 默认安装的是python2版本的, 需要卸载scons, apt autoremove scons, 安装py3的, pip3 install scons, 就可以使用scons编译了

感谢，我也是apt安装的scons不行。按你的方法，用pip3安装scons就可以了。