RT-Thread/rt-thread #3490: stall packet while receiving read_capacities command( USBmass storage)

egbert-h, 2020-03-30:
HI 

I implemented usb mass storage device with a formatted medium and ran into trouble in windows 10.   
Command handling in the mstorage.c line 614 will stall the IN endpoint if data_reside is not zero.
a shown in below figure
[image]

Remove the line 614 and the mass storage seems okay! 
[image]

[image]

anyidea? 

many thanks.

qz721, 2020-03-30:
Thanks, your issue has been noticed, we're trying to figure out what is happening.

balanceTWK, 2020-12-25:
https://github.com/RT-Thread/rt-thread/commit/0bfb881168c0c69d8d5903391c0a16d13e393ec8

This issue has been fixed by the commit above. If in doubt, you can open the issue again.

closed 2020-12-25 by balanceTWK