RT-Thread/rt-thread #3493: Bug in _read_capacity implementation

egbert-h, 2020-03-31:
HI 

SFUD driver reads the address out of bound due to the wrong value sent by _read_capacity to host.
[image]

test code & solution( mstorage.c)
[image]

[image]

tyustli, 2020-04-28:
> HI
> 
> SFUD driver reads the address out of bound due to the wrong value sent by _read_capacity to host.
> [image]
> 
> test code & solution( mstorage.c)
> [image]
> 
> [image]

thanks for your feedback.
this is a bug.
capacity = (last logical block address + 1) * block_size.
so when usb device return sector count  must descend 1.if possible you can pull request to fix this bug.thank you again.

BernardXiong, 2020-05-01:
Fix with #3569. Thank you for your contribution.

closed 2020-05-01 by BernardXiong