RT-Thread/rt-thread #3507: USB masstorage: error while receiving INQUERY command

egbert-h, 2020-04-08:
HI,

I get an error message: "xfer_len > data size" like below figure.

[image]

It seems that the function, _cb_len_calc() ,in mstorage.c  does not handle the command length "12".
[image]

USB device may stall the command while receiving the command  which the length is not "6" or "10".  
[image]

By adding handling for command length 12 in function _cb_len_calc() can workaround this issue. 
[image]

tyustli, 2020-05-07:
Thanks for your feedback.

this is a bug.when the command length is 0x0c it doesn't handle

your code can fix this.but your code only handle the COUNT case.if possible you can complete your code and pull request to fix this bug.

thank you again.

balanceTWK, 2020-12-28:
This problem has been repaired in 0bfb881168c0c69d8d5903391c0a16d13e393ec8 ， can now be closed.

closed 2020-12-28 by balanceTWK