RT-Thread/rt-thread #3508: rt_malloc设计成不能在中断里调用是出于什么考虑

rabbitsaviola, 2020-04-09:
rt_malloc使用信号量避免重入，但这样就造成无法中断里调用内存分配函数，而实际使用中通常是在中断里需要创建一条消息发送给mailbox，thread里读取mailbox的消息进行相应处理，消息由于是变长的，需要申请内存，但因为malloc无法在中断上下文使用，就无法这么实现，一个变通的办法是使用信号量或者消息队列（message queue），在ISR里释放信号量或者给message queue发消息，但如果消息是变长的，就会造成message queue空间的浪费，而用信号量又限制了线程无法判断不同触发条件。
请教rt-thread是出于什么考虑，不提供在中断上下文可以调用的内存分配函数？

BernardXiong, 2020-04-09:
rt_malloc是时间不确定的方式，不能在中断中使用，否则造成中断服务变得完全不可以预估了。

rabbitsaviola, 2020-04-11:
如果使用slab算法，malloc的时间相对就比较确定了吧？虽然这么设计能避免开发者犯错，但也限制了软件设计的自由度，作为基础软件是不是可以提供更多的灵活性，由使用者选择适合自己方案的策略，虽说是实时系统，不同软件方案对中断的响应时间的要求并不相同，比如有DMA和硬件FIFO，而且在小内存下，malloc执行时间的波动并不会很大，通过软件设计也能避免过度依赖于对中断的实时响应。

BernardXiong, 2020-04-12:
当使用slab算法时依然会存在这样的问题。如果确实希望这样做，可以从mempool中分配出来，这个是时间恒定的。

因为这个是ISR，是中断中操作，所以还是从严来对待。

rabbitsaviola, 2020-04-18:
了解了，谢谢

closed 2020-04-18 by rabbitsaviola

shao7936626, 2020-04-18:
个人理解，可以参考Linux中断的context的概念，中断的上半部分去响应中断就可以了，处理中断放到下文中；这边中断中应该是可以发一个信号出来