RT-Thread/rt-thread #3515: eth和at一起使用的时候，不接网线上电，默认网卡为eth0此时网络不可用

ethanDu1, 2020-04-10:
sal下eth和at一起配置使用时，网口不接网线上电，默认网卡会设置成eth，但是因为没接网线，此时网络不可用，但此时at是联网状态，默认网卡切换策略是否可以修改以下呢让有联网能力的网卡做默认网卡

ethanDu1, 2020-04-10:
#ifdef NETDEV_USING_AUTO_DEFAULT
/* Change to the first link_up network interface device automatically */
static void netdev_auto_change_default(struct netdev *netdev)
{
    struct netdev *new_netdev = RT_NULL;

    if (rt_memcmp(netdev, netdev_default, sizeof(struct netdev)) == 0)
    {
        new_netdev = netdev_get_first_by_flags(NETDEV_FLAG_LINK_UP);
        if (new_netdev)
        {
            netdev_set_default(new_netdev);
        }
    }
}
#endif /* NETDEV_USING_AUTO_DEFAULT */
只有默认网卡自己的状态发生改变才能从有联网能力的网卡中配置默认网卡，那一开始默认网卡就没有联网能力，状态也不发生改变那设备不就一直没有联网能力了吗？

Lawlieta, 2020-04-15:
@ethanDu1 自动切换网卡开启之后，网卡设置 up 或者 link_up 时都会检查更新一次默认网卡，如果 AT 网卡正常上网，按理说应该调用默认网卡切换操作，检查你 AT Device 中设备网卡状态设置有没有正确执行：

[image]

ethanDu1, 2020-04-16:
> @ethanDu1 自动切换网卡开启之后，网卡设置 up 或者 link_up 时都会检查更新一次默认网卡，如果 AT 网卡正常上网，按理说应该调用默认网卡切换操作，检查你 AT Device 中设备网卡状态设置有没有正确执行：
> 
> [image]

网卡设置up和link up确实会切换默认网卡，但是在切换网卡时会检查当前状态发生改变的网卡与默认网卡是否是同一个，如果不是同一个就不会切换，if (rt_memcmp(netdev, netdev_default, sizeof(struct netdev)) == 0) 这个检查是否有必要？
#ifdef NETDEV_USING_AUTO_DEFAULT
/* Change to the first link_up network interface device automatically */
static void netdev_auto_change_default(struct netdev *netdev)
{
struct netdev *new_netdev = RT_NULL;

if (rt_memcmp(netdev, netdev_default, sizeof(struct netdev)) == 0)
{
    new_netdev = netdev_get_first_by_flags(NETDEV_FLAG_LINK_UP);
    if (new_netdev)
    {
        netdev_set_default(new_netdev);
    }
}
}

ethanDu1, 2020-04-16:
> @ethanDu1 自动切换网卡开启之后，网卡设置 up 或者 link_up 时都会检查更新一次默认网卡，如果 AT 网卡正常上网，按理说应该调用默认网卡切换操作，检查你 AT Device 中设备网卡状态设置有没有正确执行：
> 
> [image]

[image]
而且netdev_low_level_set_link_status这个接口，当网卡状态UP的时候是走不到设置默认网卡的分支的吧？走的是else分支，意味着只有网卡状态down的时候，才会切换默认网卡。而不接网线上电的时候，eth是一直down的状态，at是up的状态，所以默认网卡一直是eth,只有在eth口插上网线，再拔掉，eth的状态从UP变成down才会触发切换默认网口的接口，默认网卡变成at

Lawlieta, 2020-04-16:
这个网卡的判断还是有必要的，主要用于多网卡环境下，非默认网卡状态改变时不会影响到当前默认网卡，比如现在三个网卡： ETH1（UP）、ETH2（UP）（Defaule）、AT1（UP），这时候 AT1 网卡从 UP 状态改变成 DOWN 状态，会调用该自动切换函数，如果不加当前网卡判断，默认网卡会从 ETH1 改变到 ETH2，这样的网卡改变逻辑不是我们说期望的，肯定是有问题的。

然后就是 netdev_low_level_set_link_status 接口中的判断，目前官方文档中已经指明，该网卡切换功能解释从 UP 状态到 DOWN 状态改变时会判断切换，详见下面描述：

[image]

这个状态的切换不适合放在 DOWN 到 UP 状态切换的流程中，这样也会造成上述默认网卡切换问题，所以我建议是应用层先手动检查一下默认网卡状态，或者你这边提供什么更好的方法或建议？@ethanDu1

ethanDu1, 2020-04-16:
> 这个网卡的判断还是有必要的，主要用于多网卡环境下，非默认网卡状态改变时不会影响到当前默认网卡，比如现在三个网卡： ETH1（UP）、ETH2（UP）（Defaule）、AT1（UP），这时候 AT1 网卡从 UP 状态改变成 DOWN 状态，会调用该自动切换函数，如果不加当前网卡判断，默认网卡会从 ETH1 改变到 ETH2，这样的网卡改变逻辑不是我们说期望的，肯定是有问题的。
>
> 然后就是 [...]
当前的问题场景是eth0 （link down default）,at1 (link up)。eth0网卡link状态一直是down的，at1从down初始化成up状态，而多网卡只有从UP切换到Down状态时，才会切换默认网卡，因此默认网卡一直是eth0，而eth0一直是down状态，网络不可用。能不能在初始化流程中再加一个步骤，所有网卡初始化完成后，从里面挑选一个有联网能力的网卡当默认网卡，后续默认网卡切换策略不变，防止这种多网卡场景下，初始默认网卡不可用的情况。

ethanDu1, 2020-04-16:
> 这个网卡的判断还是有必要的，主要用于多网卡环境下，非默认网卡状态改变时不会影响到当前默认网卡，比如现在三个网卡： ETH1（UP）、ETH2（UP）（Defaule）、AT1（UP），这时候 AT1 网卡从 UP 状态改变成 DOWN 状态，会调用该自动切换函数，如果不加当前网卡判断，默认网卡会从 ETH1 改变到 ETH2，这样的网卡改变逻辑不是我们说期望的，肯定是有问题的。
>
> 然后就是 [...]
我觉得放在sal层处理完比较好，在应用层还要再手动检测设置默认网卡，影响sal软件包开箱即用的效果

Lawlieta, 2020-06-01:
不建议 SAL 层涉及网卡相关的操作

oucaijun, 2020-06-19:
SAL和应用程序之间做一个网络管理层可以不？

closed 2022-03-25 by mysterywolf