RT-Thread/rt-thread #3531: 数组太大为什么会出错呢？SCB_CFSR_BFSR:0x82 PRECISERR SCB->BFAR

SuWeipeng, 2020-04-14:
如果调大下面代码中的 AP_BUFFER_MAX_SIZE 就会在运行中出错，为什么呢？
---
出错不一定在什么时候，是在运行中出错。

使用的是 STM32F407VE，master 分支内核。

```cpp
using namespace rtthread;

static uint8_t ap_buffer[AP_BUFFER_MAX_SIZE];

AP_Buffer *AP_Buffer::_instance;

AP_Buffer::AP_Buffer()
: _backend(NULL)
{
  _buf._buffer = ap_buffer;
  _buf.count = 0;
  _instance = this;
}
```

出错信息如下：
```
msh />psr: 0x61000000                                                                                
r00: 0x4134b6a6                                                                                      
r01: 0x00001fbc                                                                                      
r02: 0x00000951                                                                                      
r03: 0x4134b6a6                                                                                      
r04: 0xdeadbeef                                                                                      
r05: 0xdeadbeef                                                                                      
r06: 0xdeadbeef                                                                                      
r07: 0x2000fd30                                                                                      
r08: 0xdeadbeef                                                                                      
r09: 0xdeadbeef                                                                                      
r10: 0xdeadbeef                                                                                      
r11: 0xdeadbeef                                                                                      
r12: 0x00000000                                                                                      
 lr: 0x08005843                                                                                      
 pc: 0x0800293e                                                                                      
hard fault on thread: file                                                                           
                                                                                                     
thread   pri  status      sp     stack size max used left tick  error                                
-------- ---  ------- ---------- ----------  ------  ---------- ---                                  
loop       5  suspend 0x000000b4 0x00000800    48%   0x00000009 000                                  
file      30  running 0x000000cc 0x00001000    24%   0x00000002 000                                  
log        2  suspend 0x000000e0 0x00000400    30%   0x00000014 000                                  
led       30  suspend 0x000000a4 0x00000100    64%   0x00000014 000                                  
nrf_mav    3  suspend 0x0000038c 0x00000600    75%   0x00000001 000                                  
tshell     1  suspend 0x000000cc 0x00001000    05%   0x00000009 000                                  
vcom      16  suspend 0x0000013c 0x00000200    61%   0x00000014 000                                  
usbd       8  suspend 0x000000c0 0x00001000    04%   0x00000014 000                                  
mmcsd_de  22  suspend 0x000000b0 0x00000400    61%   0x0000000b 000                                  
tidle0    31  ready   0x00000084 0x00000200    29%   0x0000001f 000                                  
timer      2  suspend 0x000000ac 0x00000800    08%   0x00000009 000                                  
bus fault:                                                                                           
SCB_CFSR_BFSR:0x82 PRECISERR SCB->BFAR:4134B6A6 
```

SuWeipeng, 2020-04-14:
AP_BUFFER_MAX_SIZE 是 10kB 的时候出错比较频繁。

SuWeipeng, 2020-04-14:
这种问题不太好定位，能有个经验性的答复我就很满足了。

BernardXiong, 2020-04-15:
这个和你的代码相关，这些信息是看不出来原因的

closed 2020-04-15 by BernardXiong