RT-Thread/rt-thread #3539: sd card write lagre file crash

luofuchong, 2020-04-16:
hw: stm32f407
sw: webnet 2.x upload sample

when using wenet upload sample to upload file to sd card(mounted to /, add webnet、webnet/upload folder and so on), it crash when upload file is large(tested 2kB or 14kB file ):

msh />webnet_test
[I/wn] RT-Thread webnet package (V2.0.0) initialize success.
msh />Upload FileName: param
Content-Type   : application/octet-stream
save to: /webnet/upload/param
write: length 1579
psr: 0x21000000
r00: 0x20001180
r01: 0x20015f43
r02: 0x000005e0
r03: 0xffffffff
r04: 0x20015b6c
r05: 0x20015b60
r06: 0x20011b68
r07: 0x20011bd0
r08: 0x00000600
r09: 0x00000019
r10: 0x20015f43
r11: 0x00000000
r12: 0x08028479
 lr: 0x0802c5cd
 pc: 0x08000d92
hard fault on thread: webnet

thread   pri  status      sp     stack size max used left tick  error
-------- ---  ------- ---------- ----------  ------  ---------- ---
webnet    20  running 0x0000048c 0x00001000    30%   0x00000005 000
tshell    20  suspend 0x0000013c 0x00001000    08%   0x00000007 000
sys_work  23  suspend 0x0000005c 0x00000800    54%   0x0000000a 000
tcpip     10  suspend 0x000000cc 0x00000800    33%   0x00000013 000
etx       12  suspend 0x00000094 0x00000400    14%   0x00000010 000
erx       12  suspend 0x00000094 0x00000400    56%   0x0000000f 000
mmcsd_de  22  suspend 0x0000009c 0x00000400    55%   0x0000000a 000
tidle0    31  ready   0x00000054 0x00000400    08%   0x00000003 000
main      10  suspend 0x0000008c 0x00000800    44%   0x0000000d 000
usage fault:
SCB_CFSR_UFSR:0x100 UNALIGNED 

Add on;
1、If upload 1kB file is ok
2、If i use spi flash(4MB) instead of sd card, it will be no problem
3、I try to change "RT_DFS_ELM_MAX_SECTOR_SIZE" to 512/4096, it face the same problem.

BernardXiong, 2020-04-16:
Whether the memory alignment issue in SD/MMC driver in STM32F407?

luofuchong, 2020-04-17:
Thanks for your reply, I suspect it's the sdio driver problem too(not the webnet&fs components ), i will check the sdio configure of CubeMX.

hacklinshell, 2020-05-07:
> Whether the memory alignment issue in SD/MMC driver in STM32F407?

我有读文件遇到内存对齐问题，可以帮忙查看下么，谢谢

https://github.com/RT-Thread/rt-thread/issues/3583

mysterywolf, 2021-02-06:
https://github.com/RT-Thread/rt-thread/pull/4015

closed 2021-02-06 by mysterywolf