RT-Thread/rt-thread #3552: Crash issue on RCC of stm32f4.

morninf, 2020-04-21:
rt-thread version: v4.0.2.
I tested sample project of stm32f407 compiled by gcc and debugged by gdb.(arm-none-eabi-)
and I found system crash at line 884 on FILE: ./bsp/stm32/libraries/STM32F4xx_HAL/STM32F4xx_HAL_Driver/Src/stm32f400_hal_rcc.c

the code is:
pllvco = (uint32_t) ((((uint64_t) HSE_VALUE * ((uint64_t) ((RCC->PLLCFGR & RCC_PLLCFGR_PLLN) >> RCC_PLLCFGR_PLLN_Pos)))) / (uint64_t)pllm);

than I modified to:
pllvco = ((uint32_t)(((uint64_t) HSE_VALUE * ((uint64_t) ((RCC->PLLCFGR & RCC_PLLCFGR_PLLN) >> RCC_PLLCFGR_PLLN_Pos)))) / (uint32_t)pllm);

the problem is done.

someone had met this problem?

morninf, 2020-04-21:
modify to :
pllvco = (uint32_t) ((((uint32_t) HSE_VALUE * ((uint32_t) ((RCC->PLLCFGR & RCC_PLLCFGR_PLLN) >> RCC_PLLCFGR_PLLN_Pos)))) / (uint64_t)pllm);

will be ok too.

BernardXiong, 2020-04-24:
Is it caused by overflow? @SummerLife , please check it. Thanks.

SummerLife, 2020-05-06:
I often use this BSP in MDK and gcc environments and have not encountered this problem, could it be caused by some compilation parameters? @morninf

closed 2020-05-18 by morninf