RT-Thread/rt-thread #3559: 是否有计划支持C++11标准

ZhouBox, 2020-04-24:
是否有计划支持C++11标准?

BernardXiong, 2020-04-24:
Yes, but it need time to implementation.

djndl1, 2020-04-27:
Most C++11 features are already enabled by default if you're using GCC since they are not related to the standard library. The main missing part is thread support, which I am working on. I'm not sure if atomics really works or not on all architectures and with all compilers since they are probably implemented inside the compiler. I tried using atomics with G++ and did see a few extra instructions generated. Also, `std::chrono::steady_clock` seems to only have a resolution of second level since it's compiled against newlib, which has no high resolution clock.

ZhouBox, 2020-04-28:
@djndl1 Thank you. I have a library that uses std::thread std::condition_variable, but now I can't migrate it to RT.Can you share your current work？

djndl1, 2020-04-28:
> @djndl1 Thank you. I have a library that uses std::thread std::condition_variable, but now I can't migrate it to RT.Can you share your current work？

[See here](https://github.com/djndl1/rt-thread/tree/master/components/cplusplus)
`<thread>`, `<mutex>` and `<condition_variable>` are currently partly available as some classes cannot be implemented for RT-Thread because of lacking certain API. 

The thread support library currently consists of `thread`, `mutex`, `condition_variable`, `__utils.h` and their implementation files starting with `cpp_`.

ZhouBox, 2020-04-28:
@djndl1 nice!

ZhouBox, 2020-04-29:
@djndl1 main thread no call pthread create, so when call std::this_thread::get_id() no get thread id. I modify code.
```c++
pthread_t pthread_self (void)
{
    rt_thread_t tid;
    _pthread_data_t *ptd;

    tid = rt_thread_self();
    if (tid == NULL) return PTHREAD_NUM_MAX;

   // here
    return (pthread_t)tid;

    /* get pthread data from user data of thread */
    ptd = (_pthread_data_t *)rt_thread_self()->user_data;
    RT_ASSERT(ptd != RT_NULL);

    return _pthread_data_get_pth(ptd);
}

```

my event loop lib work nice;
```cpp
#include <iostream>
#include <thread>

#include "moos_includes.h"

class App : public nccloud::MoosObject 
{
    public:
        void sayHello()
        {
            std::cout << "Say Hello!" << std::endl;
        }

};

int main(void)
{
    std::cout << "Hello rt-thread" << std::endl;

    App app;

    nccloud::MoosTimer timer;
    MOOS_CONNECT(timer.timeout,&app,&App::sayHello);

    timer.start(1000);

    return app.exec();
}

```
[image]

@djndl1 Thank you very much!!!!

djndl1, 2020-05-02:
> @djndl1 main thread no call pthread create, so when call std::this_thread::get_id() no get thread id. I modify code.
> 
> ```c++
> pthread_t pthread_self (void)
> {
>     rt_thread_t tid;
>     _pthread_data_t *ptd;
> 
>     tid = rt_thread_self();
>     if (tid == NULL) return PTHREAD_NUM_MAX;
> 
>    // here
>     return (pthread_t)tid;
> 
>     /* get pthread data from user data of thread */
>     ptd = (_pthread_data_t *)rt_thread_self()->user_data;
>     RT_ASSERT(ptd != RT_NULL);
> 
>     return _pthread_data_get_pth(ptd);
> }
> ```
> 
> my event loop lib work nice;
> 
> ```c++
> #include <iostream>
> #include <thread>
> 
> #include "moos_includes.h"
> 
> 
> class App : public nccloud::MoosObject 
> {
>     public:
>         void sayHello()
>         {
>             std::cout << "Say Hello!" << std::endl;
>         }
> 
> };
> 
> int main(void)
> {
>     std::cout << "Hello rt-thread" << std::endl;
> 
>     App app;
> 
>     nccloud::MoosTimer timer;
>     MOOS_CONNECT(timer.timeout,&app,&App::sayHello);
> 
>     timer.start(1000);
> 
> 
>     return app.exec();
> }
> ```
> 
> [image]
> 
> @djndl1 Thank you very much!!!!

Yeah, the problem is that this thread support is based on pthreads, and RT-Thread has different Thread ID spaces for pthreads and native threads.

closed 2020-05-06 by ZhouBox