RT-Thread/rt-thread #3571: BSP/K210 windows下无法编译的问题

asd7893361, 2020-04-29:
下载最新版本的BSP，使用RISC-V 64bit toolchain for Kendryte K210_win32，直接编译失败，显示和signal.h中函数重定义，修改注释掉后，会出现硬浮点和软浮点冲突的问题……
有没有遇到相同问题的给解答一下呢？熊大有空看看这个问题哈

BernardXiong, 2020-05-01:
Please provide building log.

asd7893361, 2020-05-01:
the problem have been solved! the main reason is the wrong GCC tools, the GCC tool provided by Kendryte is not conform to this project, the appropriate tool can down from: https://github.com/ilg-archived/riscv-none-gcc/releases/tag/v8.2.0-2.2-20190521. and we also need to modify the "signal.h" to slove the redefinition problem.

ub3132003, 2020-05-12:
使用 none gcc  依然冲突， signal.h如何修改

asd7893361, 2020-05-12:
> 使用 none gcc 依然冲突， signal.h如何修改
把编译链中重复的函数注释掉就可以了

balanceTWK, 2020-11-20:
> 使用 none gcc 依然冲突， signal.h如何修改

我这边发现的现象是 scons -c 清理后，再编译一次，就可以了。

balanceTWK, 2020-11-20:
我这边使用 none gcc 编译正常。 @BernardXiong  这个 issue 应该可以关闭了

closed 2021-02-06 by mysterywolf