RT-Thread/rt-thread #3599: 能提供一份关于allwinner_tina的PWM的驱动代码吗？

akinggw, 2020-05-14:
能提供一份关于allwinner_tina的PWM的驱动代码吗？我看其它的bsp都有PWM的驱动代码，而allwinner_tina没有，不知道是否能官方添加一份，不甚感激。

BernardXiong, 2020-05-14:
可以根据寄存器来编写相关的驱动，欢迎提交PR

akinggw, 2020-05-15:
我从XBOOT上搞了一份PWM的驱动代码，但注册设备时老是出错：
int32_t pwm_f1c100s_init(pwm_t * pwm)
{
	write32(pwm->virt + PWM_CTRL, read32(pwm->virt + PWM_CTRL) &~(0x3fff << (pwm->channel * 15)));
	rt_kprintf("pwm_f1c100s_init:%d %d\n",pwm->virt,pwm->channel);
	write32(pwm->virt + PWM_PERIOD(pwm->channel), 0);

	//pwm_f1c100s_config(pwm, pwm->duty, pwm->period, pwm->polarity);
	//pwm_f1c100s_enable(pwm);
	return 0;
}
pwm->virt的位置是0x01c21000，pwm->channel是1，看全志的文档，应该是符合的。
出错位置在
write32(pwm->virt + PWM_PERIOD(pwm->channel), 0);
报错：

initialize rt_hw_pwm_initpwm_f1c100s_init:29495296 1
Execption:
r00:0x00003fff r01:0x01c21000 r02:0x00000000 r03:0x00000000
r04:0x8025bfbc r05:0x01c21000 r06:0xdeadbeef r07:0xdeadbeef
r08:0xdeadbeef r09:0xdeadbeef r10:0xdeadbeef
fp :0xdeadbeef ip :0x8005a8bc
sp :0x8012c7a4 lr :0x8000d768 pc :0x8000d768
cpsr:0x60000013
undefined instruction
thread - main stack:
thread   pri  status      sp     stack size max used left tick  error
-------- ---  ------- ---------- ----------  ------  ---------- ---
mmcsd_de  22  ready   0x00000040 0x00000400    06%   0x00000014 000
tidle0    31  ready   0x00000040 0x00000100    25%   0x00000020 000
timer      4  suspend 0x00000070 0x00000200    21%   0x00000009 000
main      10  running 0x00000040 0x00000800    17%   0x0000000c 000
我看全志的文档，应该是没有问题的，找不到问题了，郁闷。

BernardXiong, 2020-05-15:
加log一步步调试吧，这个在这类处理器上用log调是比较通常的模式

akinggw, 2020-05-15:
可以初始了，开始看其它bsp用的INIT_DEVICE_EXPORT，发现一直初始不了，后来看看门狗的用的INIT_BOARD_EXPORT，竟然就可以了，对rtt还是不太熟啊

closed 2021-02-05 by mysterywolf