RT-Thread/rt-thread #3630: [SFUD_SFDP] read_sfdp_data issue if SFUD_USING_QSPI enabled.

wosayttn, 2020-05-27:
We got the read_sfdp_data issue if SFUD_USING_QSPI enabled.
Its byte order of address should as below. 

[read_sfdp_data](https://github.com/RT-Thread/rt-thread/blob/master/components/drivers/spi/sfud/src/sfud_sfdp.c#L340)
->
[rt_qspi_send_then_recv](https://github.com/RT-Thread/rt-thread/blob/master/components/drivers/spi/qspi_core.c#L128)

[Solution]
static sfud_err read_sfdp_data(const sfud_flash *flash, uint32_t addr, uint8_t *read_buf, size_t size) {
    uint8_t cmd[] = {
            SFUD_CMD_READ_SFDP_REGISTER,
**#if defined(SFUD_USING_QSPI)
            (addr >> 0) & 0xFF,
            (addr >> 8) & 0xFF,
            (addr >> 16) & 0xFF,
#else**
            (addr >> 16) & 0xFF,
            (addr >> 8) & 0xFF,
            (addr >> 0) & 0xFF,
**#endif**
            SFUD_DUMMY_DATA,
    };
    SFUD_ASSERT(flash);
    SFUD_ASSERT(addr < 1L << 24);
    SFUD_ASSERT(read_buf);
    SFUD_ASSERT(flash->spi.wr);
    return flash->spi.wr(&flash->spi, cmd, sizeof(cmd), read_buf, size);
}

BernardXiong, 2020-05-29:
@armink

chrome777, 2020-05-29:
we look forward to more detailed descriptions, such as: why we should change it, or what 's the problems you have encountered in using. thanks

wosayttn, 2020-06-01:
Hi,
Current:
|  Reverse order in these caller function | Content of address of SFDP basic header |
| -- | -- |
| read_basic_header | 0x000008 |
| read_sfdp_data | 0x080000  |
| rt_qspi_send_then_recv | 0x000008 |
| **HAL_QSPI_Command(Need reverse)** | 0x080000 |

Proposed:
|  Reverse order in these caller function | Content of address of SFDP basic header |
| -- | -- |
| read_basic_header | 0x000008 |
| read_sfdp_data | 0x000008 |
| rt_qspi_send_then_recv | 0x080000 |
| **Other_hal** (send directly without reverse) | 0x080000 |

closed 2020-06-17 by wosayttn