RT-Thread/rt-thread #3635: LWIP建立UDP连接发生内存泄漏

EtcFly, 2020-05-29:
使用LWIP建立UDP连接时,  无论时数据发生还是接受, 每一次都会发生内存泄漏,  而建立TCP连接时不存在泄漏现象。测试代码如下:
```
	int sock;
	struct sockaddr_in  to;
	struct hostent *host = RT_NULL;

	host = gethostbyname("192.168.13.118");

	to.sin_family = AF_INET;
	to.sin_port = htons(6110);
	to.sin_addr = *(struct in_addr *)host->h_addr;
	rt_memset(&(to.sin_zero), 0, sizeof(to.sin_zero));

	sock = socket(AF_INET, SOCK_DGRAM, 0);
	RT_ASSERT(sock >= 0);

	uint8_t buffer[128] = "helloworld!";
       while (1)
      {
            sendto(sock, buffer, rt_strlen((void *)buffer), 0, (struct sockaddr *)&to, sizeof(struct sockaddr));
             rt_thread_mdelay(50);
      }
```
[image]

Lawlieta, 2020-05-29:
socket 创建没关闭、gethostbyname 函数第一次调用会创建全局变量

EtcFly, 2020-05-29:
> socket 创建没关闭、gethostbyname 函数第一次调用会创建全局变量

但是while循环之前的所有操作仅仅执行一次, 也就是上电后第一次会被执行, 其他时间应该一致处于循环发生状态, 即使没有关闭Socket也不会出现这种情况吧！！！

BernardXiong, 2020-05-29:
这些是需要全面来看待的，需要把问题，现象描述清楚。具体在哪步泄漏了，现象是什么，通过什么看到这个现象的。

前面lawlieta说socket没关闭，这个也是基于你提供的代码片段看到的，如果给出的信息不清晰，这些肯定会变成片面的。

EtcFly, 2020-05-29:
> 这些是需要全面来看待的，需要把问题，现象描述清楚。具体在哪步泄漏了，现象是什么，通过什么看到这个现象的。
> 
> 前面lawlieta说socket没关闭，这个也是基于你提供的代码片段看到的，如果给出的信息不清晰，这些肯定会变成片面的。

好的  如下就是一个比较完整的测试demo了, 
```
 int main(void)
{
	int sock;
	struct sockaddr_in  to;
	struct hostent *host = RT_NULL;

	host = gethostbyname("192.168.13.118");

	to.sin_family = AF_INET;
	to.sin_port = htons(6110);
	to.sin_addr = *(struct in_addr *)host->h_addr;
	rt_memset(&(to.sin_zero), 0, sizeof(to.sin_zero));

	sock = socket(AF_INET, SOCK_DGRAM, 0);
	RT_ASSERT(sock >= 0);

	uint8_t buffer[128] = "helloworld!";
       while (1)
      {
            sendto(sock, buffer, rt_strlen((void *)buffer), 0, (struct sockaddr *)&to, sizeof(struct sockaddr));
             rt_thread_mdelay(50);
      }
}
```
现象：随着scoket不断的发送数据, 通过free命令查看堆在不停的消耗,  并且没有被释放。这个过程中没有其他任何操作。

补充: 测试发现, 如果在循环中进行socket数据接受同样也会导致堆的消耗， 这个现象是测试官方tftp服务的时候发现的，当开发板作为tftp客户端，pc作为服务端时，当传输的文件较大，会由于堆内存消耗殆尽而导致传输出错

EtcFly, 2020-05-29:
> 好的 如下就是一个比较完整的测试demo了,
>
> ```
>  int main(void)
> {
> 	int sock;
> 	struct sockaddr_in  to;
> 	struct hostent *host = RT_NULL;
>
> 	host = gethostbyname("192.168.13.118");
>
> 	to.sin_family = AF_INET;
> 	to.sin_port = [...]

Lawlieta, 2020-05-29:
实际用你提供的代码测试没有出现内存泄露的现象，测试环境 PersimM3，下面是截图
[image]

请再次确定一下是否是工程中其他情况导致内存泄露，或者换一个干净的环境测试，如QEMU

EtcFly, 2020-06-01:
> 实际用你提供的代码测试没有出现内存泄露的现象，测试环境 PersimM3，下面是截图
> [image]
> 
> 请再次确定一下是否是工程中其他情况导致内存泄露，或者换一个干净的环境测试，如QEMU

对比您的测试结果，我已经发现了问题所在，1、rtt官方包和我本地包不一致，官方更新里面有涉及socket释放的操作，而我本地没有，更新同步即可。2、我的网卡移植存在内存泄漏，这也是问题的关键性，但是tcp尽然没有出现 ，费解  感谢作者

closed 2020-06-01 by EtcFly