RT-Thread/rt-thread #3639: Missing address indirection in usb device

mubes, 2020-05-31:
Hi,

In `components/drivers/usb/usbdevice/core/core.c` in `_ep0_out_notify` around line 888 the line that currently reads;

`size = dcd_ep_read(device->dcd, EP0_OUT_ADDR, ep0->request.buffer);`

should be updated to read;

`size = dcd_ep_read(device->dcd, EP0_OUT_ADDR, &ep0->request.buffer);`

...otherwise bad things happen, but not very often.

DAVE

BernardXiong, 2020-05-31:
@lymzzyh could you please confirm this issue?

lymzzyh, 2020-12-28:
我不是看的很懂。

lymzzyh, 2020-12-28:
我不是看的很懂
```c
struct uio_request
{
    rt_list_t list;
    UIO_REQUEST_TYPE req_type;
    rt_uint8_t* buffer;
    rt_size_t size;
    rt_size_t remain_size;
};
typedef struct uio_request* uio_request_t;
```

buffer本身就是个 (rt_uint8_t *)

而

```c
rt_inline rt_size_t dcd_ep_read(udcd_t dcd, rt_uint8_t address, void *buffer)
```
的入口参数是 (void *) 是出于什么理由给buffer做取址运算的？

lymzzyh, 2020-12-28:
而且目前
dcd_ep_read 并没有在任何bsp的驱动中被具体实现，所以你发生的任何问题都不会跟这玩意有关

mysterywolf, 2022-04-11:
@lymzzyh 不要在没有判定对方国籍的情况下就直接用中文回复啊