RT-Thread/rt-thread #3640: [Cortex-M23] Got hardfault issue after executing exception return in pendsv_exit.

wosayttn, 2020-06-01:
Hi,

We are porting our MCU with CM23 core to RTT system. 
Now, we got the hard fault issue in first pendsv_exit call.

We need your supporting on this issue.
Why 'bx 0xFFFFFFFC' will get the hard fault, but 'bx 0xFFFFFFFD' not?
The real exception return call from ISR should be 'bx 0xFFFFFFFD', right?

pendsv_exit
    ; restore interrupt
    MSR     PRIMASK, r2

-    MOVS    r0, #0x04
+   MOVS    r0, #0x03
    RSBS    r0, r0, #0x00
    BX      r0
    ENDP

BernardXiong, 2020-06-01:
Hi,
Could you please provide the source file and line number?

wosayttn, 2020-06-01:
Here,
[- MOVS r0, #0x04
](https://github.com/RT-Thread/rt-thread/blob/master/libcpu/arm/cortex-m23/context_rvds.S#L146)

aozima, 2020-06-17:
这里应该是 `bx 0xFFFFFFFD`.
Here should be  `bx 0xFFFFFFFD`.

```asm
    ORR     lr, lr, #0x04  ; return to thread mode and use POP
    BX      lr
    ENDP
```

> Table 2-17 Exception return behavior
> 
> EXC_RETURN[31:0] Description
> 
> 0xFFFFFFF1 Return to Handler mode, exception return uses non-floating-point state
> from the MSP and execution uses MSP after return.
> 
> 0xFFFFFFF9 Return to Thread mode, exception return uses non-floating-point state from
> MSP and execution uses MSP after return.
> 
> 0xFFFFFFFD Return to Thread mode, exception return uses non-floating-point state from
> the PSP and execution uses PSP after return.
> 
> 0xFFFFFFE1 Return to Handler mode, exception return uses floating-point-state from
> MSP and execution uses MSP after return.
> 
> 0xFFFFFFE9 Return to Thread mode, exception return uses floating-point state from
> MSP and execution uses MSP after return.
> 
> 0xFFFFFFED Return to Thread mode, exception return uses floating-point state from PSP
> and execution uses PSP after return.

@xuzhuoyi

wosayttn, 2020-06-18:
Hi,

Thanks for your reply.

We used 'bx 0xFFFFFFFC' to leave the routine, then we did some test case based on the modification. The result is unstable.
How to proof reliability on the task-scheduling and HW INT using the CM23-porting for RTT kernel? Any test suite?

I also followed your hint to test the exception return, but it can't pass compiling for CM23.
```
    ORR     lr, lr, #0x04  ; return to thread mode and use POP
    BX      lr
    ENDP
```

aozima, 2020-06-18:
>but it can't pass compiling for CM23.

Pls show you error log.

aozima, 2020-06-18:
ref #3701

closed 2020-07-01 by aozima