RT-Thread/rt-thread #3641: 睡眠模式选择可能不正确

rabbitsaviola, 2020-06-01:
https://github.com/RT-Thread/rt-thread/blob/7c761516aa24272e44f1371264096a90be41dd5e/components/drivers/pm/pm.c#L131

`_pm_change_sleep_mode`收到的睡眠模式是在开中断情况下选择出来的，有可能选择之后在关中断前别中断打断切到别的线程，改变了睡眠模式，这样是否会导致睡眠流程出错？需要依靠注册pm device规避吗？

zhangsz0516, 2020-11-09:
@rabbitsaviola 

解释如下：
执行睡眠模式的切换：如下：

```
    /* Low Power Mode Processing */
    mode = _pm_select_sleep_mode(&_pm);
    _pm_change_sleep_mode(&_pm, mode);
````
_pm_select_sleep_mode  选择模式（中断打断可能会影响模式的选择，但一旦选择了，mode不会被更改，顺序执行，外部没有更改mode的路径）

中断可以打断_pm_change_sleep_mode的执行，延时执行，但总能执行下去。

所以不用考虑睡眠流程出错问题！！

rabbitsaviola, 2020-11-14:
> @rabbitsaviola
>
> 解释如下：
> 执行睡眠模式的切换：如下：
>
> ```
>     /* Low Power Mode Processing */
>     mode = _pm_select_sleep_mode(&_pm);
>     _pm_change_sleep_mode(&_pm, mode);
> ```
>
> _pm_select_sleep_mode [...]

mode是局部变量，即使被中断打断切走，回来后还是原来的值，能继续按这个模式睡眠 ，但如果中断的处理修改了sleep_mode表，又不想睡了，但回到idle线程还会接着睡，要等到唤醒中断来才能醒过来，比如睡下去的过程中来了按键中断，本来应该不睡的，结果因为把中断处理了，还是 睡了下去

zhangsz0516, 2020-11-16:
> mode是局部变量，即使被中断打断切走，回来后还是原来的值，能继续按这个模式睡眠 ，但如果中断的处理修改了sleep_mode表，又不想睡了，但回到idle线程还会接着睡，要等到唤醒中断来才能醒过来，比如睡下去的过程中来了按键中断，本来应该不睡的，结果因为把中断处理了，还是 睡了下去

这里其实是临界值的变更问题：
* 若mode sleect 使用全局的中断保护，中断正好在选择好模式后进来。可能这个中断就丢掉了。
idle线程，是一个最低优先级的线程，只能见缝插针式的工作下去。模式选择完，如果被改掉？岂不是又要重新选择一次，依次循环。PM认定，如果模式选择了，就要执行了，中间可以做一些事情（中断、线程，都可以且走做事情！）。
* 你是否有更好的解决思路？

zhangsz0516, 2020-11-23:
@rabbitsaviola 若有好的更改思路或建议，可以提一下。

zhangsz0516, 2020-11-23:
优化方法：
https://github.com/RT-Thread/rt-thread/pull/4078

rabbitsaviola, 2020-11-26:
> 优化方法：
> #4078

抱歉回复晚了，这个修改只能降低概率，但进了_pm_change_sleep_mode之后，因为刚开始没有关中断，还是有同样的问题吧？
这里如果先关中断再选择sleep_mode进_pm_change_sleep_mode会有问题吗，假如选择是NONE，可以直接开中断退出，如果其它模式，意味着会配置唤醒源，即使中断关着，也能唤醒系统退出睡眠的，比如关闭中断准备睡眠期间，来了GPIO中断，虽然中断关着，还是能解除WFI，继续往下走开始唤醒流程的，最后执行完pm_notify.notify可以再开中断

zhangsz0516, 2020-11-26:
> > 优化方法：
> > #4078
> 
> 抱歉回复晚了，这个修改只能降低概率，但进了_pm_change_sleep_mode之后，因为刚开始没有关中断，还是有同样的问题吧？
> 这里如果先关中断再选择sleep_mode进_pm_change_sleep_mode会有问题吗，假如选择是NONE，可以直接开中断退出，如果其它模式，意味着会配置唤醒源，即使中断关着，也能唤醒系统退出睡眠的，比如关闭中断准备睡眠期间，来了GPIO中断，虽然中断关着，还是能解除WFI，继续往下走开始唤醒流程的，最后执行完pm_notify.notify可以再开中断

全局关中断，选择模式，进入睡眠，在 关中断->睡眠前处理 这个期间（很短），来了中断，中断是无法响应的，丢中断。

rabbitsaviola, 2020-11-26:
> 全局关中断，选择模式，进入睡眠，在 关中断->睡眠前处理 这个期间（很短），来了中断，中断是无法响应的，丢中断。

“在 关中断->睡眠前处理 这个期间（很短），来了中断，中断是无法响应的” ，这个是对的，但是中断状态不会清，实际并没有丢，而WFI会因为有这个pending中断继续往下走，并不会等在哪里，开了primask之后，自然会进相应的中断服务程序

zhangsz0516, 2020-11-26:
> “在 关中断->睡眠前处理 这个期间（很短），来了中断，中断是无法响应的” ，这个是对的，但是中断状态不会清，实际并没有丢，而WFI会因为有这个pending中断继续往下走，并不会等在哪里，开了primask之后，自然会进相应的中断服务程序

```
    if (mode == PM_SLEEP_MODE_NONE)
    {
        pm->sleep_mode = mode;
        pm->ops->sleep(pm, PM_SLEEP_MODE_NONE);
    }
    else
    {
        level = rt_pm_enter_critical(mode);
```
如何复现你的问题？

rabbitsaviola, 2020-11-26:
我实测时没有遇到，只是理论分析:-)

zhangsz0516, 2020-11-26:
在执行
```
   if (mode == PM_SLEEP_MODE_NONE) 
```
  这条指令时，被打断，并且功耗优先级 高于mode，会产生你的问题。这个问题不算严重，后面PM2.0上线时修复。

rabbitsaviola, 2020-11-26:
> 在执行
> 
> ```
>    if (mode == PM_SLEEP_MODE_NONE) 
> ```
> 
> 这条指令时，被打断，并且功耗优先级 高于mode，会产生你的问题。这个问题不算严重，后面PM2.0上线时修复。

PM2.0什么时候会出来啊？
我还有个想法是修改sleep_mode选择的方法，因为现在是通过调用request来选择，但实际上在用户线程里判断睡眠模式并不是很准确，用户根据什么准则来选择睡眠模式呢？通常都是空闲时间越长选择越深的睡眠模式，是不是可以在idle里线程里根据os timer的最近超时时间，选择对应的睡眠模式？

zhangsz0516, 2020-11-26:
PM2.0 正在Demo测试中，功耗相关的 文档 ：https://club.rt-thread.org/ask/article/2282.html   。有问题可以去论坛讨论。github有时不太好用。

根据空闲时间的睡眠模式，有些rtos在用，但不清楚效果如何。你可以研究下，提个pr上来。
实际使用中，tickless + deepsleep，基本能解决遇到的功耗问题。

rabbitsaviola, 2020-11-26:
> PM2.0 正在Demo测试中，功耗相关的 文档 ：https://club.rt-thread.org/ask/article/2282.html 。有问题可以去论坛讨论。github有时不太好用。
> 
> 根据空闲时间的睡眠模式，有些rtos在用，但不清楚效果如何。你可以研究下，提个pr上来。
> 实际使用中，tickless + deepsleep，基本能解决遇到的功耗问题。

好的，多谢

zhangsz0516, 2020-11-26:
这个issue建议关闭吧，你提出的问题，已经极难复现。使用中有问题，rt去论坛。

closed 2020-12-01 by rabbitsaviola