RT-Thread/rt-thread #3648: F1C100S USH HOST

Hoel, 2020-06-03:
Hello,
Is there any direction on how to add the USB HOST support for F1C100S? I have the Sunxi low level drivers working in bare metal (usbc_dev.c, usbc_dma.c, usbc_host.c, usbc_i.h, usbc_phy.c, usbc_resource.c, usbc.h, usbc.c) but i dont clearly see what has to be done to use the RTTHread USB HS HOST core.
Thank you.

BernardXiong, 2020-06-04:
USB Host or USB Device? Mostly, your bare metal code are USB device.

Hoel, 2020-06-04:
Hello, we want to implement USB HOST. The bare metal code i have is extracted from u-boot, it is the official USB stack from sunxi, and it does cover both HOST and DEVICE mode (as well as DMA) but at low level only. To implement device from this code is simple, and there is no need for a higher level stack, however that is not the case for HOST mode.

Hoel, 2020-06-04:
[F1C100S_USB_Driver_V1_1 2.zip](https://github.com/RT-Thread/rt-thread/files/4730121/F1C100S_USB_Driver_V1_1.2.zip)

This is the bare metal Eclipse project containing the files for low level driver files from Sunxi USB.

Guozhanxin, 2020-12-28:
You can refer to the STM32 USB host driver https://github.com/RT-Thread/rt-thread/blob/master/bsp/stm32/libraries/HAL_Drivers/drv_usbh.c

thinpv, 2021-03-15:
> [F1C100S_USB_Driver_V1_1 2.zip](https://github.com/RT-Thread/rt-thread/files/4730121/F1C100S_USB_Driver_V1_1.2.zip)
> 
> This is the bare metal Eclipse project containing the low level driver files from USB Sunxi.
I am investigating this matter. Have you tried the usb host feature yet?

closed 2024-12-15 by Rbb666