RT-Thread/rt-thread #3652: stm32g070cb stop0唤醒问题

rgw5267, 2020-06-04:
开启workqueue就无法唤醒，workqueue没有写任何应用。可复现

BernardXiong, 2020-06-05:
不开启stop0，只把workqueue打开是否正确？

rgw5267, 2020-06-08:
会有其他问题。
#3355 Question: STM32G070 Flash CFGBSY set by create system workqueue

zhangsz0516, 2020-11-09:
> 开启workqueue就无法唤醒，workqueue没有写任何应用。可复现
（1）是否使用PM框组件？

（2）唤醒，一般由硬件中断实现，STOP0唤醒后，时钟可能需要重新配置，所以中断处理部分，尽量不要处理太多的东西！唤醒恢复（最好有中断保护）要保证完成后，才开始用户的业务。

（3）你说的无法唤醒？指的是什么？MCU无法中断？唤醒后，马上再次进入睡眠？
（4）FLASH问题，多半是因为时钟恢复异常（ST内部的电压域可能在进入STOP模式会更改），看一下唤醒后的时钟配置。

希望你把问题描述的更清楚些，如使用的BSP，开的组件，如何唤醒（按键？其他？），这样更利于问题的排查

zhangsz0516, 2020-11-09:
@rgw5267 若在PM 功耗管理上有问题，建议去rt-thread 发帖，更能快速的回复，github网络不稳定。

rgw5267, 2020-11-10:
@zhangsz0516 感谢回复。没有使用PM组件，是自己使用的问题，后来学习了PM，低功耗管理要在idle线程中进行。现在没问题了。
#3355 是另外一个问题，和低功耗无关，因为自己不用workqueue，关闭解决问题就没有再查找根因

closed 2020-11-10 by rgw5267