RT-Thread/rt-thread #3664: Compilation problem with custom USB HOST class driver

Hoel, 2020-06-06:
I am writing a custom USH HOST class driver and i got some strange problems during compilation, it seems that the class has no visibility on usb_common.h but i wrote it exactly in accordance to mass.c which works fine and compile with no error, so there must be some special built sript/ dependency script issue that i am not aware of.
Here is the class : 

`#include <rtthread.h>
#include <drivers/usb_host.h>
#include "drivers/usb_common.h"
#include "rtl28.h"

static struct uclass_driver rtl28_driver;

#define CTRL_IN		(USB_REQ_TYPE_ENDPOINT | USB_REQ_TYPE_DIR_IN)
#define CTRL_OUT	(USB_REQ_TYPE_ENDPOINT | USB_REQ_TYPE_DIR_OUT)

uint16_t dev_read_reg(struct uintf* intf, uint8_t block, uint16_t addr, uint8_t len) {

    struct urequest setup;
    uinst_t device;
    int timeout = USB_TIMEOUT_BASIC;
    unsigned char data[2];
    uint16_t index = (block << 8);

    setup.request_type = CTRL_IN;
    setup.request = 0;
    setup.index = index;
    setup.length = len;
    setup.value = addr;

    if(rt_usb_hcd_control_xfer(device->hcd, device, &setup, (void*)data, len, timeout) != 0)
        rt_kprintf("dev_read_reg error\n");

    reg = (data[1] << 8) | data[0];
    return reg;
}
rt_err_t dev_write_reg(struct uintf* intf, uint8_t block, uint16_t addr, uint16_t val, uint8_t len) {

    struct urequest setup;
    uinst_t device;
    int timeout = USB_TIMEOUT_BASIC;
    unsigned char data[2];
    uint16_t index = (block << 8) | 0x10;

    if (len == 1){ data[0] = val & 0xff; }
    else{ data[0] = val >> 8; }
    data[1] = val & 0xff;

    setup.request_type = CTRL_OUT;
    setup.request = 0;
    setup.index = index;
    setup.length = len;
    setup.value = 0;

    if(rt_usb_hcd_control_xfer(device->hcd, device, &setup, (void*)data, len, timeout) == 0) return RT_EOK;
    else return -RT_FALSE;
}

static rt_err_t rt_usbh_rtl28_enable(void* arg){

}
static rt_err_t rt_usbh_rtl28_disable(void* arg){

}
ucd_t rt_usbh_class_driver_rtl28(void){

}`

and here is the compilation output : 

hoel@iMac-de-Hoel-7 ~/Downloads/rt-thread-master/bsp/allwinner_tina_prj_v1 $ scons
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
scons: building associated VariantDir targets: build
CC build/kernel/components/drivers/usb/usbhost/class/rtl28.o
/Users/hoel/Downloads/rt-thread-master/components/drivers/usb/usbhost/class/rtl28.c:11:30: warning: 'struct uintf' declared inside parameter list
 uint16_t dev_read_reg(struct uintf* intf, uint8_t block, uint16_t addr, uint8_t len) {
                              ^
/Users/hoel/Downloads/rt-thread-master/components/drivers/usb/usbhost/class/rtl28.c:11:30: warning: its scope is only this definition or declaration, which is probably not what you want
/Users/hoel/Downloads/rt-thread-master/components/drivers/usb/usbhost/class/rtl28.c: In function 'dev_read_reg':
/Users/hoel/Downloads/rt-thread-master/components/drivers/usb/usbhost/class/rtl28.c:20:10: error: 'struct urequest' has no member named 'request'
     setup.request = 0;
          ^
/Users/hoel/Downloads/rt-thread-master/components/drivers/usb/usbhost/class/rtl28.c:21:10: error: 'struct urequest' has no member named 'index'
     setup.index = index;
          ^
/Users/hoel/Downloads/rt-thread-master/components/drivers/usb/usbhost/class/rtl28.c:22:10: error: 'struct urequest' has no member named 'length'
     setup.length = len;
          ^
/Users/hoel/Downloads/rt-thread-master/components/drivers/usb/usbhost/class/rtl28.c:23:10: error: 'struct urequest' has no member named 'value'
     setup.value = addr;
          ^
/Users/hoel/Downloads/rt-thread-master/components/drivers/usb/usbhost/class/rtl28.c:25:8: warning: implicit declaration of function 'rt_usb_hcd_control_xfer' [-Wimplicit-function-declaration]
     if(rt_usb_hcd_control_xfer(device->hcd, device, &setup, (void*)data, len, timeout) != 0)
        ^
/Users/hoel/Downloads/rt-thread-master/components/drivers/usb/usbhost/class/rtl28.c:28:5: error: 'reg' undeclared (first use in this function)
     reg = (data[1] << 8) | data[0];
     ^
/Users/hoel/Downloads/rt-thread-master/components/drivers/usb/usbhost/class/rtl28.c:28:5: note: each undeclared identifier is reported only once for each function it appears in
/Users/hoel/Downloads/rt-thread-master/components/drivers/usb/usbhost/class/rtl28.c: At top level:
/Users/hoel/Downloads/rt-thread-master/components/drivers/usb/usbhost/class/rtl28.c:31:31: warning: 'struct uintf' declared inside parameter list
 rt_err_t dev_write_reg(struct uintf* intf, uint8_t block, uint16_t addr, uint16_t val, uint8_t len) {
                               ^
/Users/hoel/Downloads/rt-thread-master/components/drivers/usb/usbhost/class/rtl28.c: In function 'dev_write_reg':
/Users/hoel/Downloads/rt-thread-master/components/drivers/usb/usbhost/class/rtl28.c:44:10: error: 'struct urequest' has no member named 'request'
     setup.request = 0;
          ^
/Users/hoel/Downloads/rt-thread-master/components/drivers/usb/usbhost/class/rtl28.c:45:10: error: 'struct urequest' has no member named 'index'
     setup.index = index;
          ^
/Users/hoel/Downloads/rt-thread-master/components/drivers/usb/usbhost/class/rtl28.c:46:10: error: 'struct urequest' has no member named 'length'
     setup.length = len;
          ^
/Users/hoel/Downloads/rt-thread-master/components/drivers/usb/usbhost/class/rtl28.c:47:10: error: 'struct urequest' has no member named 'value'
     setup.value = 0;
          ^
/Users/hoel/Downloads/rt-thread-master/components/drivers/usb/usbhost/class/rtl28.c: In function 'rt_usbh_rtl28_enable':
/Users/hoel/Downloads/rt-thread-master/components/drivers/usb/usbhost/class/rtl28.c:55:1: warning: no return statement in function returning non-void [-Wreturn-type]
 }
 ^
/Users/hoel/Downloads/rt-thread-master/components/drivers/usb/usbhost/class/rtl28.c: In function 'rt_usbh_rtl28_disable':
/Users/hoel/Downloads/rt-thread-master/components/drivers/usb/usbhost/class/rtl28.c:58:1: warning: no return statement in function returning non-void [-Wreturn-type]
 }
 ^
/Users/hoel/Downloads/rt-thread-master/components/drivers/usb/usbhost/class/rtl28.c: At top level:
/Users/hoel/Downloads/rt-thread-master/components/drivers/usb/usbhost/class/rtl28.c:53:17: warning: 'rt_usbh_rtl28_enable' defined but not used [-Wunused-function]
 static rt_err_t rt_usbh_rtl28_enable(void* arg){
                 ^
/Users/hoel/Downloads/rt-thread-master/components/drivers/usb/usbhost/class/rtl28.c:56:17: warning: 'rt_usbh_rtl28_disable' defined but not used [-Wunused-function]
 static rt_err_t rt_usbh_rtl28_disable(void* arg){
                 ^
/Users/hoel/Downloads/rt-thread-master/components/drivers/usb/usbhost/class/rtl28.c:6:29: warning: 'rtl28_driver' defined but not used [-Wunused-variable]
 static struct uclass_driver rtl28_driver;
                             ^
/Users/hoel/Downloads/rt-thread-master/components/drivers/usb/usbhost/class/rtl28.c: In function 'dev_read_reg':
/Users/hoel/Downloads/rt-thread-master/components/drivers/usb/usbhost/class/rtl28.c:30:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
/Users/hoel/Downloads/rt-thread-master/components/drivers/usb/usbhost/class/rtl28.c: In function 'rt_usbh_class_driver_rtl28':
/Users/hoel/Downloads/rt-thread-master/components/drivers/usb/usbhost/class/rtl28.c:61:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
scons: *** [build/kernel/components/drivers/usb/usbhost/class/rtl28.o] Error 1
scons: building terminated because of errors.

What is the problem?

BernardXiong, 2020-06-08:
I think it should be `struct uhintf`.

balanceTWK, 2020-12-28:
Since there has been no response to this issue for a long time, the issue will be closed now. If necessary, you can open it again!

closed 2020-12-28 by balanceTWK