RT-Thread/rt-thread #3672: stm32软件i2c问题

zxc199108, 2020-06-09:
文件：drv_soft_i2c.c

这个软件模拟的i2c不够标准，在进行接收时，没有将sda与scl的引脚改为输入模式。当然，i2c一般都会上拉电阻，提高驱动能力，这样设置为od模式的模拟i2c还能使用。但是在某些场景下(i2c线过长、上拉电阻问题)，会因驱动能力不够导致i2c异常。例如我用stm32的官方板子，测试i2c到i2c的通信是可以的，但是rt的软件i2c是无法使用的，因为没有上拉电阻。

这是我修改的代码，这种方式无上拉电阻也可使用，比较接近硬件i2c：

static void stm32_i2c_gpio_init(struct stm32_i2c *i2c)
{
    struct stm32_soft_i2c_config* cfg = (struct stm32_soft_i2c_config*)i2c->ops.data;

    rt_pin_mode(cfg->scl, PIN_MODE_OUTPUT); //add by zxc
    rt_pin_mode(cfg->sda, PIN_MODE_OUTPUT); //add by zxc 
	

    rt_pin_write(cfg->scl, PIN_HIGH);
    rt_pin_write(cfg->sda, PIN_HIGH);
}

/**
 * This function sets the sda pin.
 *
 * @param Stm32 config class.
 * @param The sda pin state.
 */
static void stm32_set_sda(void *data, rt_int32_t state)
{
    struct stm32_soft_i2c_config* cfg = (struct stm32_soft_i2c_config*)data;
		rt_pin_mode(cfg->sda, PIN_MODE_OUTPUT); //add by zxc
    if (state)
    {
			
        rt_pin_write(cfg->sda, PIN_HIGH);
    }
    else
    {
        rt_pin_write(cfg->sda, PIN_LOW);

    }
}

/**
 * This function sets the scl pin.
 *
 * @param Stm32 config class.
 * @param The scl pin state.
 */
static void stm32_set_scl(void *data, rt_int32_t state)
{
    struct stm32_soft_i2c_config* cfg = (struct stm32_soft_i2c_config*)data;

		rt_pin_mode(cfg->scl, PIN_MODE_OUTPUT); //add by zxc
    if (state)
    {
        rt_pin_write(cfg->scl, PIN_HIGH);
				
    }
    else
    {
				
        rt_pin_write(cfg->scl, PIN_LOW);
    }
}

/**
 * This function gets the sda pin state.
 *
 * @param The sda pin state.
 */
static rt_int32_t stm32_get_sda(void *data)
{
    struct stm32_soft_i2c_config* cfg = (struct stm32_soft_i2c_config*)data;
		rt_pin_mode(cfg->sda, PIN_MODE_INPUT_PULLUP); //add by zxc
    return rt_pin_read(cfg->sda);
}

/**
 * This function gets the scl pin state.
 *
 * @param The scl pin state.
 */
static rt_int32_t stm32_get_scl(void *data)
{
    struct stm32_soft_i2c_config* cfg = (struct stm32_soft_i2c_config*)data;
		rt_pin_mode(cfg->scl, PIN_MODE_INPUT_PULLUP); //add by zxc
    return rt_pin_read(cfg->scl);
}

BernardXiong, 2020-06-10:
Welcome to submit your PR, thanks.

z14git, 2020-06-25:
标准就是要开漏结构吧😕

zxc199108, 2020-07-02:
> 标准就是要开漏结构吧

嗯，硬件标准确实是，我就是发现像MicroPython或者是一些开发板商的模拟I2C是内部上拉的，包括我自己以前使用过1米长的I2C，发现改硬件上拉电阻 过于麻烦，就内部改一下引脚模式就能用了，RT的不也是虚拟I2C。当然，开漏结构没有任何问题，毕竟是标准。

Connorzengbohao, 2020-07-19:
切换输入输出的时候会产生短暂的电平上下拉（用逻辑分析仪可以抓取到），有些i2c通讯的芯片会识别到，导致通信失败呀。

zxc199108, 2020-07-20:
> 切换输入输出的时候会产生短暂的电平上下拉（用逻辑分析仪可以抓取到），有些i2c通讯的芯片会识别到，导致通信失败呀。

是么？我之前用逻辑分析仪测试输入输出切换的软件I2C时，并没有看到短暂的电平翻转，改天用示波器抓一下看看

closed 2020-07-20 by zxc199108