RT-Thread/rt-thread #3698: 关于ffs内联汇编在ARMCLANG下开启LTO的BUG

geniusgogo, 2020-06-17:
最近在ARMCLANG编译器下开了LTO选项，发现一个内联汇编的BUG。具体错误是链接时，发生了全局符号冲突。
```
linking...
.\build\keil\Obj\UHFDoor.axf: error: invalid symbol redefinition
exit:                        
.\build\keil\Obj\UHFDoor.axf: ERROR: Error parsing inline asm
".\build\keil\Obj\UHFDoor.axf" - 2 Error(s), 0 Warning(s).
Target not created.
```
https://github.com/RT-Thread/rt-thread/blob/d7d6dbe774e6d497a235984b9247f1aa4a86d83a/libcpu/arm/cortex-m3/cpuport.c#L397

试着将exit符号改成局部符号后链接成功，不知道这样改是否有其它问题，请大佬过目：
```
#elif defined(__CLANG_ARM)
int __rt_ffs(int value)
{
    __asm volatile(
        "CMP     r0, #0x00            \n"
        "BEQ     1f                   \n"

        "RBIT    r0, r0               \n"
        "CLZ     r0, r0               \n"
        "ADDS    r0, r0, #0x01        \n"

        "1:                           \n"

        : "=r"(value)
        : "r"(value)
    );
    return value;
}
```

geniusgogo, 2020-06-17:
这个错误的原因通过反向验证，是因为LTO优化开起来后会将__rt_ffs函数多次进行inline，导致了__asm块中的exit符号被多次redefine。

BernardXiong, 2020-06-18:
Thank you for your feedback and PR.

closed 2020-06-18 by geniusgogo