RT-Thread/rt-thread #3713: tm4c129x rt_application_init allocate error

orhaneee, 2020-06-22:
Hi,

In tm4c129x, I got an error
`(tid != RT_NULL) assertion failed at function:rt_application_init, line number:200`

I have tried it all compilers and the result is same.

BernardXiong, 2020-06-22:
Please check your memory heap setting. It's caused by out of memory normally.

orhaneee, 2020-06-22:
Yes, on .icf file it has been set to 0x0. I tried to change it to some value and built again but the result is same. I also tried enabling debug options especially memory and it says no memory as you specified.

BernardXiong, 2020-06-23:
If you have JTAG or other ICE-link, you can debug it step by step, especially in memory heap settings.

orhaneee, 2020-06-23:
Do you have another recommendation for this? What about heap size in linker file? I don’t understand how it cannot allocate heap

ArdaFu, 2020-06-24:
1. Here is the heap definition: [tm4c129x/applications/board.h](https://github.com/RT-Thread/rt-thread/blob/master/bsp/tm4c129x/applications/board.h) line 26 to 36. As the code all unused SRAM has been defined as HEAP.
2. Please advice your building environment. e.g. toolchain, RT-Thread OS version. 
3. I suggest you attach your minimum system code that we can help you to trace the problem.

ArdaFu, 2020-06-24:
Hi, as Bernard said, the automatic components will not initialized the HEAP. We should manually do that at rt_hw_board_init function. I've committed the PR, hope that can help your problem. @orhaneee

orhaneee, 2020-06-24:
@ArdaFu Hi Arda, with your PR the issue was resolved. Thank you!

closed 2020-06-24 by orhaneee