RT-Thread/rt-thread #3720: [FINSH]__CC_ARM宏定义在AC6下未定义

jiladahe1997, 2020-06-24:
我的开发环境：
_STM32CubeMX + Keil + Keil run-time RT-thread nano(包含finsh)_

问题描述：
STM32CubeMX默认生成的Keil工程编译器是ARM Compiler 5，编译一次大约需要2-3分钟左右，影响开发速度。 于是尝试切换到ARM Compiler 6，切换之后，发现__CC_ARM宏未定义，导致了FINSH无法使用，如下图所示。

[image]

我去ARM官方文档找了下，发现了ARM Compiler 6确实不包含这个宏定义：
ARM Compiler6包含的宏定义：http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dui0742g/chr1398241769674.html
ARM Compiler5包含的宏定义：http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dui0472k/chr1359125007083.html

需要帮忙解答的问题：
       1.为什么要判断__CC_ARM才能定义FINSH的相关宏定义，是因为FINSH不支持gcc编译器吗？
       2.根据上述我找到的文档，AC6不包含__CC_ARM宏定义，我个人猜想FINSH应该是支持AC6的，能不能直接把`#if defined(__CC_ARM) || defined(__CLANG_ARM)`这行注释掉呢？

yangjie11, 2020-06-24:
感谢反馈，你可以先删除这个条件。这里后面修一下

geniusgogo, 2020-07-03:
难道AC6没有__CLANG_ARM吗？

mysterywolf, 2021-06-17:
该问题在nano3.1.5版本中已经解决

closed 2021-06-17 by mysterywolf