RT-Thread/rt-thread #3737: SDIO框架问题

whj4674672, 2020-07-04:
thread/blob/4108836a9003c411e127eb5bea80a5d09cd61bd5/components/drivers/sdio/sdio.c#L893
这里通过宏定义CARD_FLAG_HIGHSPEED直接给定了50M的速度，这个在STM32F2上如果同时使能了USB和SDIO的情况下这个时钟最高只能是48M.
另外https://github.com/RT-Thread/rt-thread/blob/4108836a9003c411e127eb5bea80a5d09cd61bd5/components/drivers/sdio/sdio.c#L734
这里设置高速的条件是SDIO device需要设置MMCSD_SUP_HIGHSPEED，这样就又出现了高速卡的时钟是固定50M.

whj4674672, 2020-07-04:
``    /* Check if user Clock div < Normal speed 25Mhz, no change in Clockdiv */
    if(hsd->Init.ClockDiv >= SDMMC_NSpeed_CLK_DIV)
    {
      Init.ClockDiv = hsd->Init.ClockDiv;
    }
    else if (hsd->SdCard.CardSpeed == CARD_ULTRA_HIGH_SPEED)
    {
      /* UltraHigh speed SD card,user Clock div */
      Init.ClockDiv = hsd->Init.ClockDiv;
    }
    else if (hsd->SdCard.CardSpeed == CARD_HIGH_SPEED)
    {
      /* High speed SD card, Max Frequency = 50Mhz */
      Init.ClockDiv = SDMMC_HSpeed_CLK_DIV;
    }
    else
    {
      /* No High speed SD card, Max Frequency = 25Mhz */
      Init.ClockDiv = SDMMC_NSpeed_CLK_DIV;
    }

    (void)SDMMC_Init(hsd->Instance, Init);
``
这部分是ST HAL对时钟处理的方式

Guozhanxin, 2020-12-23:
修改代码已经合并了，如果没问题了的话，可以关闭此issue

closed 2021-01-02 by BernardXiong