RT-Thread/rt-thread #3756: console device支持DMA_TX模式

rabbitsaviola, 2020-07-21:
请问console device是否可以修改成支持DMA_TX模式，这样可以减少打印的CPU开销？

BernardXiong, 2020-07-22:
不行的，console设备是以字符模式发送，并且还允许在中断中打印log，所以rt_kprintf本身是没考虑到这样的情况。

如果考虑到日志的效能，可以多用LOG_D/W等，然后自己做一个log的后端，先输出到memory中，然后再通过统一的途径发送出来

rabbitsaviola, 2020-07-22:
这样一来console是不是就不能与日志（比如ulog）复用一个uart了？因为console要以polling模式使用uart，而ulog又要用DMA模式？

rabbitsaviola, 2020-07-23:
可以这样修改rt_kprintf的实现，以便将它和ulog统一吗，ulog_init_ok是增加的接口，返回ulog.init_ok
```c
#ifdef RT_USING_ULOG
if (!ulog_init_ok)
{
    _console_device->open_flag |= RT_DEVICE_FLAG_STREAM;
    /* Avoid using DMA even though uart device is configured in DMA mode.
     *  It happens in intialization stage if ULOG is enabled.
     */
    _console_device->open_flag &= ~RT_DEVICE_FLAG_DMA_TX;
    rt_device_write(_console_device, 0, rt_log_buf, length);
    _console_device->open_flag = old_flag
}
else
{
    ulog_raw("%s", rt_log_buf);
}
#else
//old code
#endif
```

BernardXiong, 2020-07-23:
可以先自行这样使用

rabbitsaviola, 2020-07-24:
请问有统一rt_kprintf和ulog的计划吗，能使用DMA毕竟效率高些，log和console复用一个串口也便于调试，console除了回显的时候是单个字符发送，使用DMA效率反而低，其它时候还是会批量发送字符的吧，另外就是在ulog异步模式打开的情况下，RT_ASSERT还需要增加flush的操作。

BernardXiong, 2020-07-26:
暂时没有，也许会和usb log一起考虑。

closed 2020-08-07 by BernardXiong