RT-Thread/rt-thread #3789: stm32h7xx的模板SConscript路径错误

ziyouchutuwenwu, 2020-07-30:
board/SConscript

```python
if rtconfig.CROSS_TOOL == 'gcc':
    src += [cwd + '/../../libraries/STM32H7xx_HAL/CMSIS/Device/ST/STM32H7xx/Source/Templates/gcc/startup_stm32h743xx.s']
elif rtconfig.CROSS_TOOL == 'keil':
    src += [cwd + '/../../libraries/STM32H7xx_HAL/CMSIS/Device/ST/STM32H7xx/Source/Templates/arm/startup_stm32h743xx.s']
elif rtconfig.CROSS_TOOL == 'iar':
    src += [cwd + '/../../libraries/STM32H7xx_HAL/CMSIS/Device/ST/STM32H7xx/Source/Templates/iar/startup_stm32h743xx.s']
```

应该使用
```python
Import('SDK_LIB')

startup_path_prefix = SDK_LIB
src += [startup_path_prefix + '/STM32H7xx_HAL/CMSIS/xxxxxxxxx'
```

不然`scons --dist` 出来的项目,复制到外面,编译提示路径找不到

SummerLife, 2020-07-30:
欢迎提交 pr 修复该问题 😃

ziyouchutuwenwu, 2020-07-30:
> 欢迎提交 pr 修复该问题  ，看起来是漏更新了 H7 系列的 BSP 模板。

不仅仅是stm32h743-st-nucleo里面的这个脚本错误, 这里的也不对
```bash
bsp/stm32/libraries/templates/stm32h7xx/board/SConscript
```

SummerLife, 2020-07-30:
> > 欢迎提交 pr 修复该问题  ，看起来是漏更新了 H7 系列的 BSP 模板。
> 
> 不仅仅是stm32h743-st-nucleo里面的这个脚本错误, 这里的也不对
> 
> ```shell
> bsp/stm32/libraries/templates/stm32h7xx/board/SConscript
> ```

h743 BSP 的修改已经被提交了，你只需要修改模板中的脚本就可以了，欢迎提交 PR 😃

SummerLife, 2020-08-01:
另外我看了一下，你说的模板中的问题，在 2020.2.25 号的 [PR](https://github.com/RT-Thread/rt-thread/pull/3402/commits/d9a9b90b33501ec715021715cb98819dd88d786b) 已经修复过了，也许你手中的 RT-Thread 不是最新版本因此才没有更新，可以看一下 master 版本，已经修复了这个问题。

SummerLife, 2020-08-01:
@BernardXiong  相关问题已经修复，该 issue 可以关闭了。

closed 2020-08-01 by BernardXiong