RT-Thread/rt-thread #3794: 建议将LWIP 2.0.2升级到2.0.3 或增设2.0.3版本

mysterywolf, 2020-07-30:
2.0.3版本要比2.1.2还要稳定。2.1.2在反复的socket-close时，快慢定时器会出现不工作的问题，导致close后无法释放内存，逐渐将内存耗光，最开始我以为是我自己的问题，后来我和其他工程师交流的时候发现他们也遇到了相同的问题，我们通过不同的方式溯源，最后都查出来是快慢定时器的问题。

BernardXiong, 2020-08-01:
可以提供份测试代码吗？

mysterywolf, 2020-08-03:
@BernardXiong  我已经上传到我的仓库里边了，请找到lwip-socket-test这个仓库。仓库中附了另一位工程师和我出现相同问题之后的分析笔记（分析的过程不一定对，但是出现的问题和结论是一致的）。

BernardXiong, 2020-08-04:
@xiangxistu 请留意、后续跟进这个问题。

maolin-china, 2020-08-07:
您好，我们在用lwip 2.0.2版本，也发现了相同的问题，请问lwip 2.0.3版本能解决该问题吗？但是在2.0.3的bugFixs中没有发现该问题的修复。

mysterywolf, 2020-08-07:
2.0.3中没有这个问题了，已经测试过了，但是在2.1.2中不知道为什么又复现了。反复socket -close这种情况用的比较少，我是在web client走http1.0协议的时候暴露出来的

maolin-china, 2020-08-08:
最终发现我们底层移植的问题，与lwip2.0.2版本没有关系，
我们应用rtt的代码时做了一些修改，
但是十分感谢！

mysterywolf, 2020-08-08:
那可能是2.0.2原始源码没有这个问题，但是在2.1.2版本中，我没有使用RT-THREAD操作系统，而是ucos操作系统，lwip也是自己移植的，最开始我是怀疑我自己移植的问题，但是后来又有工程师也反应了同样的毛病，我们的工程生成路线是完全不一样的，所以才开始怀疑是不是lwip2,.1.2源码有问题。

mysterywolf, 2020-08-08:
@maolin-china  能具体说一下是底层哪里的问题吗 避免其他人在使用2.0.2的时候也出现同样的问题

BernardXiong, 2020-08-09:
> 最终发现我们底层移植的问题，与lwip2.0.2版本没有关系，
> 我们应用rtt的代码时做了一些修改，
> 但是十分感谢！

请具体说明下情况哈

maolin-china, 2020-08-10:
@mysterywolf @BernardXiong 
我们的产品遇到的问题及定位的方法总结一下
**问题现象**
产品中设计了一个tcp server，在client反复 open/close之后，发现heap最终被耗光

**问题原因**
问题在lwip中sys_mbox_free函数，我们的工程师在sys_mbox_new中增加了一部分代码，申请了一些内存，在sys_mbox_free的时候没有全部释放。

**定位方法**
之前怀疑是lwip的问题，看了 @mysterywolf 的问题后，感觉我们的问题很相似，决定升级lwip的版本，但问题并没有解决。
决定对堆的申请和释放进行反复跟踪才定位到的这个地方。

mysterywolf, 2020-08-10:
@maolin-china 你如果采用的是RTT官方的sys_mbox_free、sys_mbox_new出现了上述问题，请你贴一下sys_mbox_free、sys_mbox_new代码，谢谢。

maolin-china, 2020-08-10:
@mysterywolf RTT官方的没有问题，我们是基于官方代码增加了一些内容，并没有什么参考意义，代码没有必要贴了。

longtengmcu, 2020-09-02:
> 2.0.3版本要比2.1.2还要稳定。2.1.2在反复的socket-close时，快慢定时器会出现不工作的问题，导致close后无法释放内存，逐渐将内存耗光，最开始我以为是我自己的问题，后来我和其他工程师交流的时候发现他们也遇到了相同的问题，我们通过不同的方式溯源，最后都查出来是快慢定时器的问题。

我也遇到过lwip 2.0.3长时间连接后在特殊的情况下断开TCP链接，导致TCP块不释放的问题，我已经自行解决，请参考我的这个文章：https://club.rt-thread.org/ask/question/424406.html

mysterywolf, 2020-09-02:
> > 2.0.3版本要比2.1.2还要稳定。2.1.2在反复的socket-close时，快慢定时器会出现不工作的问题，导致close后无法释放内存，逐渐将内存耗光，最开始我以为是我自己的问题，后来我和其他工程师交流的时候发现他们也遇到了相同的问题，我们通过不同的方式溯源，最后都查出来是快慢定时器的问题。
> 
> 我也遇到过lwip 2.0.3长时间连接后在特殊的情况下断开TCP链接，导致TCP块不释放的问题，我已经自行解决，请参考我的这个文章：https://club.rt-thread.org/ask/question/424406.html

收藏了 木五老师也说过 tcpip协议本身就有bug

BernardXiong, 2020-09-04:
😳 这个不是bug啊，TCP/IP协议本身就是这样规定的。

关闭后，不全关，先进入FIN_WAIT_1状态，等待。超时后，完全关闭掉。

如果这个时候要创建新的连接，如果发现没socket了，会从FIN_WAIT_1状态里的socket硬摘一个出来使用。lwIP基本上是这样的一个处理方式，没毛病

longtengmcu, 2020-09-04:
> 😳 这个不是bug啊，TCP/IP协议本身就是这样规定的。
> 关闭后，不全关，先进入FIN_WAIT_1状态，等待。超时后，完全关闭掉。
> 如果这个时候要创建新的连接，如果发现没socket了，会从FIN_WAIT_1状态里的socket硬摘一个出来使用。lwIP基本上是这样的一个处理方式，没毛病

从协议上来讲确实是没有毛病的，但是lwip协议在设计实现连网效果时，在我测试的一种特殊情况下，会造成tcp控制块全部被用尽，无法新分配块，并且无法创建新的链接而无法再连网，这种状态持续的时间还比较长，所以这在实际应用中是无法接受的。

mysterywolf, 2020-09-04:
> 从协议上来讲确实是没有毛病的，但是lwip协议在设计实现连网效果时，在我测试的一种特殊情况下，会造成tcp控制块全部被用尽，无法新分配块，并且无法创建新的链接而无法再连网，这种状态持续的时间还比较长，所以这在实际应用中是无法接受的。

你把2MSL设置的小一点一般来说LWIP_MSL 3000 即2MSL为6秒足够了

mysterywolf, 2021-06-28:
https://github.com/RT-Thread/rt-thread/pull/4824

closed 2021-06-28 by mysterywolf