RT-Thread/rt-thread #3800: 关于硬件定时器的线程安全问题

jiladahe1997, 2020-08-03:
## 问题发生的场景和现象：
使用rt-thread大佬写的modbus库：[https://github.com/armink/FreeModbus_Slave-Master-RTT-STM32](url)

在程序运行后有极小的几率会在串口中断中卡死。

## 问题大致定位的位置

在串口中断中会调用`rt_timer_start`来启动定时器，程序会卡死在`timer.c`下面这个for循环中：
```
/* timer.c 354-376行*/
354:         for (; row_head[row_lvl] != timer_list[row_lvl].prev;
355:              row_head[row_lvl]  = row_head[row_lvl]->next)
356:         {
357:             struct rt_timer *t;
358:             rt_list_t *p = row_head[row_lvl]->next;
359: 
360:             /* fix up the entry pointer */
361:             t = rt_list_entry(p, struct rt_timer, row[row_lvl]);
362: 
363:             /* If we have two timers that timeout at the same time, it's
364:              * preferred that the timer inserted early get called early.
365:              * So insert the new timer to the end the the some-timeout timer
366:              * list.
367:              */
368:             if ((t->timeout_tick - timer->timeout_tick) == 0)
369:             {
370:                 continue;
371:             }
372:             else if ((t->timeout_tick - timer->timeout_tick) < RT_TICK_MAX / 2)
373:             {
374:                 break;
375:             }
376:         }

```
##### 观察变量可以看到如下结果：定时器链表 **rt_timer_list** 链表变成了死循环链表：
[image]

##### 通过多次断点尝试，发现了如下的调用栈和调用位置：
[image]
[image]

## 猜测的原因和解决办法

可以看到在执行`rt_timer_start`时被打断了，打断后又执行了一遍`rt_timer_start`，导致的后果的是，执行顺序变成了：
 1. `_rt_timer_remove`   此时被打断
 2. `_rt_timer_remove` 
 3. `rt_list_insert_after` 
 4.  回到被打断的位置，继续执行：`rt_list_insert_after`

第一次remove已经移除了链表节点，第二次remove无效；第三次第四次插入两次，造成链表变成死循环链表。

#### 可能的解决办法
个人认为解决办法在于 `timer.c` 中：
```
/*timer.c 318行-336行 */
318:     /* stop timer firstly */
319:     level = rt_hw_interrupt_disable();
320:     /* remove timer from list */
321:     _rt_timer_remove(timer);
322:     /* change status of timer */
323:     timer->parent.flag &= ~RT_TIMER_FLAG_ACTIVATED;
324:     rt_hw_interrupt_enable(level);
325: 
326:     RT_OBJECT_HOOK_CALL(rt_object_take_hook, (&(timer->parent)));
327: 
328:     /*
329:      * get timeout tick,
330:      * the max timeout tick shall not great than RT_TICK_MAX/2
331:      */
332:     RT_ASSERT(timer->init_tick < RT_TICK_MAX / 2);
333:     timer->timeout_tick = rt_tick_get() + timer->init_tick;
334: 
335:     /* disable interrupt */
336:     level = rt_hw_interrupt_disable();

```
在324行手动开启了中断，之后又在336行几乎是又立即关闭了中断。所以我个人认为：

 - 324行这里是否不需要开启中断。

 - 或者另一方面来说，`rt_timer_start`这个函数本身就是线程不安全的？即使这样也希望能做一个断言判断一下，不然这样在死循环里不出来很难排查问题。

谢谢。

BernardXiong, 2020-08-07:
可否给个简单的流程？因为freemodbus的代码太多，并没去看那边的代码。

是：

* isr -> timer timeout，<一次性timer?>
* 然后再去rt_timer_start一个timer？

jiladahe1997, 2020-08-07:
> 可否给个简单的流程？因为freemodbus的代码太多，并没去看那边的代码。
> 
> 是：
> 
> * isr -> timer timeout，<一次性timer?>
> * 然后再去rt_timer_start一个timer？

画了两个简单的流程图，请参考。
 - *理想情况的流程图*：
[image]

- *实际有极小概率会出现的流程图*：
[image]

<br />
<br />

其实我个人认为，我提到的问题和modbus库本身关系不大。

我个人认为`rt_timer_start`这个函数在**多个线程中对同一个定时器使用时**，可能会被打断，造成会在定时器链表中，连续插入两次同一个定时器对象，使得链表变成死循环。

这是根据我实际遇到的情况，以及对timer.c代码324行处开启了中断的分析得出的怀疑。如果最后不对的话还请见谅。

BernardXiong, 2020-08-13:
-_- 不太懂modbus，没用过

貌似是说，多线程调用rt_timer_start出现了问题？是因为其中关闭中断是分了两次，这样有可能在中间被抢占，导致混乱了？

jiladahe1997, 2020-08-13:
> -_- 不太懂modbus，没用过
> 
> 貌似是说，多线程调用rt_timer_start出现了问题？是因为其中关闭中断是分了两次，这样有可能在中间被抢占，导致混乱了？

是的，是这个意思。

但是发生的情况比较苛刻，需要同时满足两个条件：

- 线程A的rt_timer_start被线程B打断后又执行了rt_timer_start.
- 线程A的rt_timer_start和线程B的rt_timer_start操作的是同一个timer.

BernardXiong, 2020-08-13:
收到，这个是bug点，后续进行修复。  :bug:

Guozhanxin, 2020-11-30:
我提了个pr，大佬有时间试试啊 https://github.com/RT-Thread/rt-thread/pull/4094

jiladahe1997, 2020-12-04:
我写了个demo复现这个问题。demo的工作原理如下(*由于demo只是针对stm32h743的，代码就不放上来了*)：
<br/>

[image]

测试结果：
- 按照如图流程，能够复现此bug，系统会卡死
- 按照pr中的代码进行修改后，bug消失，系统不会卡死。（**注：仅在上述流程中能表明bug修复，其他case都没有进行更多测试，建议官方进行更多测试**）

closed 2021-01-02 by BernardXiong