RT-Thread/rt-thread #3902: MDK ARMCLANG build error

geniusgogo, 2020-09-11:
../middleware/rt-thread/include\rtthread.h(24): warning: In file included from...
../middleware/rt-thread/include\rtdef.h(1094): warning: In file included from...
../middleware/rt-thread/include/rtlibc.h(15): warning: In file included from...
../middleware/rt-thread/include/libc/libc_stat.h(14): error: 'sys/stat.h' file not found

BernardXiong, 2020-09-11:
@mysterywolf  Please check them carefully.

mysterywolf, 2020-09-11:
got it 麻烦标个bug label

mysterywolf, 2020-09-11:
bug已排除 @BernardXiong  @geniusgogo

geniusgogo, 2020-09-12:
你提交在哪里了？

geniusgogo, 2020-09-12:
@mysterywolf 单独发我一份patch，我先测试一下

mysterywolf, 2020-09-12:
@geniusgogo  https://github.com/mysterywolf/rt-thread 你可以fork一份

geniusgogo, 2020-09-12:
> @geniusgogo https://github.com/mysterywolf/rt-thread 你可以fork一份

可以了，我这里编译通过

mysterywolf, 2020-09-12:
> > @geniusgogo https://github.com/mysterywolf/rt-thread 你可以fork一份
> 
> 可以了，我这里编译通过

@BernardXiong  OK了

BernardXiong, 2020-09-12:
PR呢？合并在哪？

mysterywolf, 2020-09-12:
> PR呢？合并在哪？

https://github.com/RT-Thread/rt-thread/pull/3898 这里

mysterywolf, 2020-10-11:
这个issue可以关了 @geniusgogo

closed 2020-10-14 by geniusgogo