RT-Thread/rt-thread #3910: 关于头文件相互包含导致递归编译的问题 说明

mysterywolf, 2020-09-13:
## 1. 什么是头文件“相互包含”

A头文件中包含了B头文件，B头文件中也包含了A头文件。导致两个头文件来回相互调用形成死锁，**这种问题不会立刻显现，一旦受到一些机缘巧合的文件改变（增加或减少），编译器就会开始报错**。现象为：明明在A头文件中声明了`a_uint_t`数据类型，但是在包含A头文件的`.c`文件中使用依然会报无法找到`a_uint_t`数据类型的错误。

## 2. 该问题的受影响对象

GCC编译链对这种问题并不敏感，可以正常使用。但是KEIL-MDK编译器的armcc编译链，编译要求比GCC链苛刻，会出现`error:  #20: identifier "xxxx" is undefined`的错误。

## 3. 目前在RT-Thread代码中发现的存在头文件相互包含的问题实例

### 3.1

`rt-thread\components\drivers\include\rtdevice.h`中

```c
#ifdef RT_USING_PIN
#include "drivers/pin.h"
#endif
```

而`drivers/pin.h`中又包含了<rtdevice.h>，进而导致两个头文件相互包含。

```c
#include <rtthread.h>
#include <rtdevice.h>
```

### 3.2 

`rt-thread\components\drivers\include\ipc\pipe.h` 中包含了<rtdevice.h>

```c
/**
 * Pipe Device
 */
#include <rtthread.h>
#include <rtdevice.h>
```

而在rtdevice.h中又包含了pipe.h

```c
#include "ipc/ringbuffer.h"
#include "ipc/completion.h"
#include "ipc/dataqueue.h"
#include "ipc/workqueue.h"
#include "ipc/waitqueue.h"
#include "ipc/pipe.h"  /*<-----这里*/
#include "ipc/poll.h"
#include "ipc/ringblk_buf.h"
```

等等

## 4.该问题需要引起重视，尽可能规避头文件相互包含的现象

balanceTWK, 2020-09-15:
没遇到过这种问题欸，我们那个 BSP 用 MDK 编译会报错？我去瞧瞧！！！（{PS：ARMCC 版本号 MDK 版本号再告知下！）

```c
#ifndef __RT_XXX_H__
#define __RT_XXX_H__

。。。。。。略

#endif /* __RT_XXX_H__ */
```

我感觉不会有这种问题啊，如果 `__RT_XXX_H__` 定义了，这个头文件里的内容肯定也有定义啊。如果没定义那就更好说了，直接编译里面的内容。

mysterywolf, 2020-09-15:
> 没遇到过这种问题欸，我们那个 BSP 用 MDK 编译会报错？我去瞧瞧！！！（{PS：ARMCC 版本号 MDK 版本号再告知下！）
> 
> ```c
> #ifndef __RT_XXX_H__
> #define __RT_XXX_H__
> 
> 。。。。。。略
> 
> #endif /* __RT_XXX_H__ */
> ```
> 
> 我感觉不会有这种问题啊，如果 `__RT_XXX_H__` 定义了，这个头文件里的内容肯定也有定义啊。如果没定义那就更好说了，直接编译里面的内容。

我已经提pr改过来了

BernardXiong, 2020-11-19:
好的，关注。后续一个个的来看这块的问题

closed 2022-06-17 by mysterywolf