RT-Thread/rt-thread #3912: RT-Thread编程陷阱与防坑指南清单

geniusgogo, 2020-09-14:
此issue登记记录在开发过程中遇到的一些使用陷阱，并最好能提供如何规避此类陷阱的方法、思路以及故事:) 

* rt_malloc/rt_free
> - 内核版本：<4.0.3
> - 陷阱1：不应该出现在中断上下文中
>> - 说明：由于rt_malloc/rt_free里会操作锁，RT-Thread在中断上下文中不可拿锁。

mysterywolf, 2020-10-10:
## 不要在rt-thread中使用rt_thread_suspend函数，除非你非常清楚这个函数的实现机理

如果A线程中含有rt_thread_delay rt_sem_take等函数时，在B线程中调用rt_thread_suspend企图让A线程挂起，若正好好撞到A线程正在执行rt_thread_delay rt_sem_take函数时，会导致B线程调用的rt_thread_suspend函数失效

失效原因为：
因为在rt_thread_delay rt_sem_take等阻塞函数内部也调用了rt_thread_suspend

而rt-thread并不支持嵌套挂起/解挂，在B线程中rt_thread_suspend运行到

    if ((thread->stat & RT_THREAD_STAT_MASK) != RT_THREAD_READY)
    {
        RT_DEBUG_LOG(RT_DEBUG_THREAD, ("thread suspend: thread disorder, 0x%2x\n", thread->stat));

        return -RT_ERROR;
    }
会被直接判定为失败并返回错误码

因此会导致在上述情况下rt_thread_suspend失效

在标准情况下，一个线程必须主动调用阻塞函数以让出cpu使用权，换句话说rt_thread_suspend能否执行，在标准线程设计中将变成随缘行为，如果运气好没有撞到阻塞函数，即可正常挂起；若运气不好整好撞上了阻塞函数，则无法正确挂起。

##从其他操作系统转到RTT的用户要格外注意
经过测试uCOS-III以及FreeRTOS均不存在该问题，如果用户是从上述两个操作系统迁移过来的，直接使用RT-Thread 的这个API, 在没有任何提示的情况下保持原有编程思想，将会出现问题。

KyleChenjh, 2021-01-23:
# 针对flag=RT_IPC_FLAG_FIFO的mutex使用建议 

RT-Thread 针对 `mutex`提供了两种互斥锁的阻塞等待模式：

一种是 `flag = RT_IPC_FLAG_FIFO`,一种是`flag = RT_IPC_FLAG_PRIO`。

使用`RT_IPC_FLAG_PRIO`创建的IPC对象，在多个线程等待资源时，将由优先级高的线程优先获得资源。
而使用`RT_IPC_FLAG_FIFO`则是创建的IPC对象，在多个线程等待资源时，将按照先来先得的顺序获得资源。

在使用 `RT_IPC_FLAG_FIFO` 时候，由于未进行优先级排列，如果suspend链表上的最后几个线程优先级很高，那么互斥任务在release的时候，为了防止优先级反转，会将自身优先级提升到链表的最高优先级，这就会造成 **不需要使用该互斥锁的中间优先级的任务** 要一直等待整个链表结束（因为高优先级任务在链表最后边，需要等待）才能运行。如果这个等待整个链表的时间的是不能容忍的，那么建议使用 `RT_IPC_FLAG_PRIO` 的方式进行。

mysterywolf, 2021-04-24:
# 标准C库 时间函数 <time.h>
目前，C库的时间函数已经相当完善，用户可以正常使用所有C标准库中的函数以及宏定义，建议使用最新的master版本。但是唯一的问题是，目前的C库时间函数，涉及到当地时间以及时区的问题，支持并不是很完善，目前时区固定为UTC+8（东八区，即北京时间），用户需要手动调整时区。该问题预计4.0.4版本之前不会解决，因为涉及到一个比较棘手的问题，即不同国家的夏令时时间不一致，导致时区的修正非常复杂。
手动修改时区的方式：
[image]
[image]
只需要修改这两个函数的内部utc_plus值即可，例如西2区(UTC-2)，就改为 utc_plus = -2

建议用户使用<sys/time.h>代替<time.h>，因为<time.h>仅包含C标准库的时间函数，而<sys/time.h>不仅包含C标准库相关时间函数，而且还包含Linux系统以及POSIX标准规定的相关时间函数。

mysterywolf, 2021-04-25:
#双精度浮点问题
部分STM32高级芯片具备双精度浮点计算能力，设置方式有所区别
参见：
https://github.com/RT-Thread/rt-thread/issues/4186 ->  https://github.com/RT-Thread/rt-thread/pull/4188
https://github.com/RT-Thread-Studio/sdk-bsp-stm32h750-realthread-artpi/issues/177 -> https://github.com/RT-Thread-Studio/sdk-bsp-stm32h750-realthread-artpi/pull/195

mysterywolf, 2021-05-17:
# 关于RT_IPC_FLAG_FIFO
不要随意使用RT_IPC_FLAG_FIFO来创建信号量、互斥量、消息队列以及邮箱，该flag会使用上述内核对象的线程处于非实时状态，强烈建议采用RT_IPC_FLAG_PRIO；除非您看重任务提交信号量等的先来后到，并了解RT_IPC_FLAG_FIFO可能导致相关线程实时性丧失，可以使用RT_IPC_FLAG_FIFO。

mysterywolf, 2021-07-02:
# 慎用RT_IPC_FLAG_FIFO
除非你有明确理由，非常关心内核对象等待队列排列顺序的先来后到，否则强烈建议用RT_IPC_FLAG_PRIO

haodongnj, 2021-11-24:
# STM32 CAN Bus 的比特率
STM32的CAN驱动里面代码适配时，有预设的总线时钟。需要核查驱动预设的时钟和你使用的板子的总线时钟是否匹配。如果不匹配，则需要自行修改驱动里面的时钟配置选项，否则CAN总线的bitrate会不合预期。

enkiller, 2022-02-22:
# scons Windows 平台编译，命令行参数超过 32K 报错规避办法

1. 打开工程目录下 `SConstruct` 文件：

2. 在 `objs = PrepareBuilding(env, RTT_ROOT)` 这一行前面添加如下代码

```python
def ourspawn(sh, escape, cmd, args, e):
    newargs = ' '.join(args[1:])
    cmdline = cmd + " " + newargs
    if (len(cmdline) > 16 * 1024):
        f = open('cmd.param', 'w')
        f.write(' '.join(args[1:]).replace('\\', '/'))
        f.close()
        # exec
        cmdline = cmd + " @cmd.param"
    proc = subprocess.Popen(cmdline, stdin=subprocess.PIPE, stdout=subprocess.PIPE,
        stderr=subprocess.PIPE, shell = False, env = e)
    data, err = proc.communicate()
    rv = proc.wait()
    if rv != 0:
        sys.stderr.write(err)
    else:
        sys.stdout.write(data)
    return rv

if platform.system() == 'Windows':
    env['SPAWN'] = ourspawn
```

3. 在文件头部导入相应的模块

```python
import subprocess
import platform
```

supperthomas, 2022-05-18:
[rt-thread 使用宝典](https://club.rt-thread.org/ask/article/2460fcd7db4821ae.html)