RT-Thread/rt-thread #3946: 在下述情况下rt_thread_suspend失效

mysterywolf, 2020-10-04:
如果A线程中含有rt_thread_delay rt_sem_take等函数时，在B线程中调用

rt_thread_suspend企图让A线程挂起，若正好好撞到A线程正在执行rt_thread_delay 

rt_sem_take函数时，会导致B线程调用的rt_thread_suspend函数失效

失效原因为：
因为在rt_thread_delay rt_sem_take等阻塞函数内部也调用了rt_thread_suspend

而rt-thread并不支持递归挂起/解挂，在rt_thread_suspend运行到
```c
    if ((thread->stat & RT_THREAD_STAT_MASK) != RT_THREAD_READY)
    {
        RT_DEBUG_LOG(RT_DEBUG_THREAD, ("thread suspend: thread disorder, 0x%2x

\n",
                                       thread->stat));

        return -RT_ERROR;
    }
```
会被直接判定为失败并返回错误码

因此会导致在上述情况下rt_thread_suspend失效

BernardXiong, 2020-10-04:
rt_thread_suspend在大多数场合是一个内部接口，一般用户不应该直接去调用它。当要进行多线程操作时，推荐更多以sem/mutex/event/msgq/mbox等方式进行。

mysterywolf, 2020-10-04:
我是在编写兼容层的时候发现的这个问题，经过测试uCOS-III以及FreeRTOS均不存在该问题，如果用户是从上述两个操作系统迁移过来的，在没有任何提示的情况下保持原有编程思想，将会出现问题。建议对该函数予以修改，或者在API手册中显著位置予以提示用户该函数的受限点以及受限原因。一个任务命令另一个任务挂起是一个比较常见的手段。

KyleChenjh, 2020-11-18:
首先suspend/resume没有嵌套的设计理念，我个人认为是合理的。

第一点是本身已经suspend的thread，怎么可能再被suspend?逻辑就是不通的，这个不像互斥量的设计思想，允许线程递归持有的原因是为了继承优先级。

第二点是设计上，suspend的触发条件很苛刻，即RT_THREAD_READY或者RT_THREAD_RUNNING的的线程（其实严格来说是rt_thread_self线程，需要手动rt_schedule才行），因为这样的条件才有意义。

第三点是比较关键的，就是多核的调度。线程的状态只有线程自身最清楚，状态改变靠调度器就够了，如果非要主动改变，那也是自身线程，你在多核上，A线程是完全洞察不到B线程的状态，A线程让B线程挂起本身就是不合理的。至于为什么ucos是允许递归调用suspend，是因为ucos的设计上就没考虑多核，仅仅就是单核思想，这点我觉得是RT-Thread比ucos优越的地方之一。不能仅仅为了考虑兼容，就改变整个设计理念（多核往单核上去实现？）兼容层应该可以换种思想去实现。
```c
if (stat == RT_THREAD_RUNNING)
{
    /* not suspend running status thread on other core */
    RT_ASSERT(thread == rt_thread_self());
}
```
第四点是suspend接口更多是偏向于内部系统实现，用户一般不要使用该接口，人肉运维调度是不可取的，要不然要调度器干什么，对于A线程与B线程的逻辑，完全可以用其他的方式去实现，比如信号量或者邮箱等等。

mysterywolf, 2020-11-18:
"偏向于内部系统实现" 偏向这个词过于暧昧，到底是允许用户使用还是不允许用户使用？ 问题的关键在于，在RT-Thread中，挂起解挂函数根本没法用。

KyleChenjh, 2020-11-18:
前面第四点也说了，能用该接口，但是用的时候要非常的注意和明白自己设计的系统的逻辑，清楚它的各种情况的运行状态，要不然就干脆别用，本来就有调度器干这活。当然linux在多核上面有太对的限制加了很多锁。

mysterywolf, 2020-11-18:
所以问题的关键在于这个接口有bug，用都没法用。我的这个issue阐述的主要是这个问题。

closed 2021-04-17 by mysterywolf