RT-Thread/rt-thread #3973: rt_thread_delay_until() wrong delay time

warmonkey, 2020-10-21:
What's wrong:
rt_thread_delay_until() actual delay time is much longer (1.5-2x) than expected

Test case:

Code analysis:
In file thread.c line 541 function rt_err_t rt_thread_delay_until:
`if (rt_tick_get() - *tick < inc_tick)
    {
        *tick = rt_tick_get() - *tick + inc_tick;
        /* suspend thread */
        rt_thread_suspend(thread);
        /* reset the timeout of thread timer and start it */
        rt_timer_control(&(thread->thread_timer), RT_TIMER_CTRL_SET_TIME, *tick);
        rt_timer_start(&(thread->thread_timer));
`
RT_TIMER_CTRL_SET_TIME is setting timer delay in ticks. the original propose of updating *tick is compensating execution time between the function entry and the calculation, which the elapsed time is rt_tick_get() - *tick. 
The actual delay should be inc_tick minus elapsed time, not elapsed + inc_tick time.

**How to fix**
we should set RT_TIMER_CTRL_SET_TIME  to (inc_tick-elapsed):
replace line 556-565 with:
`
    rt_tick_t elapsed = rt_tick_get() - *tick;
    if (elapsed < inc_tick)
    {
        rt_tick_t inc_set = inc_tick - elapsed ;
        /* suspend thread */
        rt_thread_suspend(thread);
        /* reset the timeout of thread timer and start it */
        rt_timer_control(&(thread->thread_timer), RT_TIMER_CTRL_SET_TIME, &inc_set);
`

mysterywolf, 2020-10-22:
You are right. In theory, LED should light on for 20ms rather than 37ms.

BernardXiong, 2020-10-22:
Thank you for you feedback. You are right.

BernardXiong, 2020-10-22:
fix with #3977. Thank you for your feedback.

closed 2020-10-25 by BernardXiong