RT-Thread/rt-thread #4018: nrf5x/nrf52840首次MDK编译失败

githubhjz, 2020-11-10:
clone代码后，使用pgks --update更新好，使用scons --target=mdk5生成工程，但是编译失败，编译错误信息如下：
`Build started: Project: project
*** Using Compiler 'V5.06 update 6 (build 750)', folder: 'D:\Program Files (x86)\Keil_v5\ARM\ARMCC\Bin'
Build target 'rtthread'
compiling drv_uart.c...
..\libraries\drivers\drv_uart.c(32): error:  #20: identifier "NRFX_UART0_INST_IDX" is undefined
      .uart = NRFX_UART_INSTANCE(0),
..\libraries\drivers\drv_uart.c: 0 warnings, 1 error
compiling board.c...
packages\nrfx-v2.1.0\drivers\include\nrfx_power_clock.h(55): error:  #35: #error directive: "This code is not supposed to be compiled when neither POWER nor CLOCK is enabled."
      #error "This code is not supposed to be compiled when neither POWER nor CLOCK is enabled."
board\board.c: 0 warnings, 1 error
".\build\rtthread.axf" - 2 Error(s), 0 Warning(s).
Target not created.
Build Time Elapsed:  00:00:03`

githubhjz, 2020-11-10:
需要通过env开启串口就可以编译通过了

closed 2020-11-10 by githubhjz

Guozhanxin, 2020-11-10:
这个也可以提个pr修复下。默认开启串口。