RT-Thread/rt-thread #4045: drv_common.c文件中的HAL_Delay函数与以太网初始化存在冲突

maxroom, 2020-11-20:
使用系统版本为：4.0.3 build Nov 17 2020
参照官方教程：https://www.rt-thread.org/document/site/application-note/components/network/an0011-network-started/
进行以太网的移植，一直不成功（报错信息未记录）

后发现是该版本中在drv_common.c中写的HAL_Delay：
void HAL_Delay(__IO uint32_t Delay)
{

rt_thread_mdelay(Delay);
}
使用了系统delay函数，但此时调度尚未启动，该函数会出错，建议解决方法：
1，以太网初始化放到下一级
2，取消drv_common.c中HAL_Delay函数的实现或改用其他方法实现

BernardXiong, 2020-11-28:
是的，一些和delay相关的初始化建议都放到任务中进行初始化，而不是在调度启动之前就初始化 @balanceTWK

balanceTWK, 2020-11-28:
> 是的，一些和delay相关的初始化建议都放到任务中进行初始化，而不是在调度启动之前就初始化 @balanceTWK

void HAL_Delay(__IO uint32_t Delay)

里面的实现使用裸机的 for 循环 可以不。 @BernardXiong

mysterywolf, 2020-11-29:
rt_thread_mdelay里边可以加一个判断，如果调度器还没有启动的话 直接调用rt_hw_us_delay死卡。初始化应该放在任务中，不建议死卡。

BernardXiong, 2020-12-07:
> > 是的，一些和delay相关的初始化建议都放到任务中进行初始化，而不是在调度启动之前就初始化 @balanceTWK
> 
> void HAL_Delay(__IO uint32_t Delay)
> 
> 里面的实现使用裸机的 for 循环 可以不。 @BernardXiong

for循环延时可以的

mysterywolf, 2020-12-31:
#4135

closed 2021-01-02 by BernardXiong