RT-Thread/rt-thread #4075: The object has been deleted. Why does the list still exist

zjlahtc, 2020-11-23:
https://github.com/RT-Thread/rt-thread/blob/ad8d95b1c4502b302f87b4f89ce9a7c9c3fcdf38/src/object.c#L498

[image]

任务里创建了名为top的定时器，再删除定时器重复操作，发现定时器列表中依然有残留信息。请帮忙检查下代码

BernardXiong, 2020-11-23:
有更进一步的描述吗？或者测试用例？单次定时器执行完毕后是需要手动删除的，并不会有自动的行为。

zjlahtc, 2020-11-23:
rt_timer_create函数关于RT_TIMER_FLAG_ONE_SHOT单次执行没有明细说明，希望添加下注释，或者任务执行完自动删除。这样更方便使用。另外也希望rt_timer_create可以添加首次delay执行的时间【rt_timer_create(rt_tick_t delay, rt_tick_t period)】，目前编写top指令时发现第一次执行定时器任务不是很方便。shell调用执行到top函数后才输出msh />，所以第一次需要延时等待“FINSH_PROMPT”输出再打屏top命令的信息，故希望是否可以考虑下这种方式。

[image]

[image]

BernardXiong, 2020-11-24:
后面在文档里提及下，自动删除的操作是不会做的。

zjlahtc, 2020-12-10:
3Q

closed 2020-12-18 by BernardXiong