RT-Thread/rt-thread #4076: rt_thread_suspend不能保证执行过程完全关中断

MenglongWu, 2020-11-23:
__rt-thread v4.0.3__  
  不能保证rt_thread_suspend执行全过程中断是关闭的，临界状态会在rt_schedule_remove_thread、rt_timer_stop函数里打开中断。

```
rt_err_t rt_thread_suspend(rt_thread_t thread)
{
    temp = rt_hw_interrupt_disable();
    rt_schedule_remove_thread(thread);
    rt_timer_stop(&(thread->thread_timer));
    rt_hw_interrupt_enable(temp);
}

void rt_schedule_remove_thread(struct rt_thread *thread)
{
    level = rt_hw_interrupt_disable();
    rt_hw_interrupt_enable(level);
}

rt_err_t rt_timer_stop(rt_timer_t timer)
{
    level = rt_hw_interrupt_disable();
    rt_hw_interrupt_enable(level);
}
```

  加上开关中断使用计数是否能更安全呢？
  内核API不直接调用 **rt_hw_interrupt_disable、rt_hw_interrupt_enable**，替代为 **_rt_hw_interrupt_disable、_rt_hw_interrupt_enable**。
   
```
register rt_base_t g_save_it = 0;
atomic_int g_it_count = 0;

rt_base_t _rt_hw_interrupt_disable(void)
{
    if (0 == count) {
        atomic_add(count);
        g_save_it = rt_hw_interrupt_disable();
        return g_save_it;
    }
    return g_save_it;
}
void _rt_hw_interrupt_enable(tmp)
{
    atomic_sub(count);
    if (0 == count) {
        temp = rt_hw_interrupt_disable();
    }
}
```

mysterywolf, 2020-11-23:
rt_hw_interrupt_enable rt_hw_interrupt_disable 本身就可以嵌套中断 
rt_schedule_remove_thread rt_timer_stop函数内部已经对临界区套中断保护了
虽然你表述有些问题 但是 不建议你使用rt_thread_suspend函数 https://github.com/RT-Thread/rt-thread/issues/3946

mysterywolf, 2020-12-30:
@balanceTWK  该issue可以关闭了

closed 2020-12-30 by balanceTWK