RT-Thread/rt-thread #4082: drv_usart.c 中STM32 设置串口参数时，关于奇偶校验位的BUG

TMODER, 2020-11-24:
stm32_configure(struct rt_serial_device *serial, struct serial_configure *cfg)中，当设置串口奇偶校验位的时候：
switch (cfg->parity)
    {
    case PARITY_NONE:
        uart->handle.Init.Parity     = UART_PARITY_NONE;
        break;
    case PARITY_ODD:
        uart->handle.Init.Parity     = UART_PARITY_ODD;
        break;
    case PARITY_EVEN:
        uart->handle.Init.Parity     = UART_PARITY_EVEN;
        break;
    default:
        uart->handle.Init.Parity     = UART_PARITY_NONE;
        break;
    }
实际使用发现，当有校验位的时候，输出始终不对。后来发现 STM32的串口当有校验位的时候，WordLength需要设置为9位。建议修改为：
    switch (cfg->parity)
    {
    case PARITY_NONE:
        uart->handle.Init.Parity     = UART_PARITY_NONE;
        break;
    case PARITY_ODD:
        uart->handle.Init.Parity     = UART_PARITY_ODD;
	**uart->handle.Init.WordLength = UART_WORDLENGTH_9B;**
        break;
    case PARITY_EVEN:
        uart->handle.Init.Parity     = UART_PARITY_EVEN;
	**uart->handle.Init.WordLength = UART_WORDLENGTH_9B;**
        break;
    default:
        uart->handle.Init.Parity     = UART_PARITY_NONE;
        break;
    }

BernardXiong, 2020-11-24:
Welcome PR, thanks.

whj4674672, 2020-12-04:
仅仅这个地方修改应该无法解决9位的问题，可以查看这个https://github.com/RT-Thread/rt-thread/issues/3341

TMODER, 2020-12-04:
> 仅仅这个地方修改应该无法解决9位的问题，可以查看这个#3341

实际测试使用是可以的：数据位设置8位，串口设置奇偶校验位是没有问题的。如果设置了奇偶位，但HAL库中的UART的WordLength不设置为UART_WORDLENGTH_9B，输入数据和输出就不一样。

closed 2021-02-25 by mysterywolf