RT-Thread/rt-thread #4095: 缺少对rt_thread_init()函数的错误检查

Hxinrong, 2020-11-27:
您好，
根据项目API手册，rt_thread_init()函数用于初始化一个线程，成功返回 RT_EOK, 失败则返回 -RT_ERROR。在下面代码中，关于thread2的rt_thread_init()函数的返回值可能缺少对“result”变量的赋值，
https://github.com/RT-Thread/rt-thread/blob/dfa99ad16546c2b1b03187e634201295606f4246/examples/kernel/thread_priority.c#L47-L67

并且，在下列代码中，可能缺少对该函数的返回值的错误检查。
https://github.com/RT-Thread/rt-thread/blob/dfa99ad16546c2b1b03187e634201295606f4246/bsp/frdm-k64f/applications/application.c#L74-L80
https://github.com/RT-Thread/rt-thread/blob/dfa99ad16546c2b1b03187e634201295606f4246/bsp/lpc178x/applications/application.c#L148-L154
https://github.com/RT-Thread/rt-thread/blob/dfa99ad16546c2b1b03187e634201295606f4246/bsp/lpc2148/applications/application.c#L154-L166
https://github.com/RT-Thread/rt-thread/blob/dfa99ad16546c2b1b03187e634201295606f4246/bsp/nuclei/hbird_eval/applications/main.c#L26-L28
https://github.com/RT-Thread/rt-thread/blob/dfa99ad16546c2b1b03187e634201295606f4246/bsp/nuvoton/libraries/nu_packages/Demo/usbd_hid_dance_mouse.c#L102-L108
https://github.com/RT-Thread/rt-thread/blob/dfa99ad16546c2b1b03187e634201295606f4246/bsp/pic32ethernet/application.c#L51-L56
https://github.com/RT-Thread/rt-thread/blob/dfa99ad16546c2b1b03187e634201295606f4246/components/drivers/usb/usbdevice/class/cdc_vcom.c#L952-L957
https://github.com/RT-Thread/rt-thread/blob/dfa99ad16546c2b1b03187e634201295606f4246/src/timer.c#L756-L766

然而在其他的一些调用位置，有检查该函数的返回值。例如，
https://github.com/RT-Thread/rt-thread/blob/dfa99ad16546c2b1b03187e634201295606f4246/bsp/apollo2/applications/main.c#L65-L76
https://github.com/RT-Thread/rt-thread/blob/dfa99ad16546c2b1b03187e634201295606f4246/bsp/at32/Libraries/rt_drivers/drv_eth.c#L751-L756
https://github.com/RT-Thread/rt-thread/blob/dfa99ad16546c2b1b03187e634201295606f4246/bsp/lpc824/applications/application.c#L71-L73

mysterywolf, 2020-12-01:
你好 rtthread是开源软件 由大家一起来维护 在你发现问题之后 欢迎给rt thread贡献代码  https://www.bilibili.com/video/BV1gr4y1w7yX

BernardXiong, 2020-12-01:
谢谢您的反馈并指出代码中的一些问题，如果可能，还希望可以多多向rt-thread提交修正代码的PR。另外，请 @Guozhanxin 来跟进后续的事项。

谢谢

Guozhanxin, 2020-12-02:
十分感谢楼主反馈这么多的问题，不知楼主有没有想法再进一步的修复这些问题呢，有问题的话可以随时联系我，我的邮箱guozhanxin@rt-thread.com😜

Hxinrong, 2020-12-07:
嗯嗯好哒

Hxinrong, 2020-12-07:
#4133 add some error checks of function rt_thread_init()

closed 2022-01-21 by mysterywolf