RT-Thread/rt-thread #4105: 缺少函数rt_event_recv()的返回值检查

Hxinrong, 2020-11-30:
您好，
函数rt_event_recv()用于接收事件，返回RT_EOK ，-RT_ETIMEOUT 或 -RT_ERROR。但是，在如下代码中，缺少对该函数是否成功执行的返回值错误检查。
https://github.com/RT-Thread/rt-thread/blob/af95348bf1dd06ed2147cf9caee8a78513c31a72/components/dfs/filesystems/jffs2/src/gcthread.c#L201-L204
https://github.com/RT-Thread/rt-thread/blob/af95348bf1dd06ed2147cf9caee8a78513c31a72/components/dfs/filesystems/jffs2/src/gcthread.c#L225-L229
https://github.com/RT-Thread/rt-thread/blob/af95348bf1dd06ed2147cf9caee8a78513c31a72/bsp/nrf5x/libraries/templates/nrf52x/applications/ble_nus_app.c#L603-L604

然而，在其他调用处都检查了该函数的返回值，如，
https://github.com/RT-Thread/rt-thread/blob/af95348bf1dd06ed2147cf9caee8a78513c31a72/bsp/stm32f20x/Drivers/i2c.c#L422
https://github.com/RT-Thread/rt-thread/blob/af95348bf1dd06ed2147cf9caee8a78513c31a72/bsp/stm32/libraries/HAL_Drivers/drv_sdio.c#L133-L139

whj4674672, 2020-12-04:
@Guozhanxin 确实存在这个问题

Hxinrong, 2020-12-07:
#4134

Hxinrong, 2020-12-07:
#4134 add error checks of function rt_event_recv()

Guozhanxin, 2020-12-09:
感谢您的贡献，我们近期就会安排人来审核您的 pr😀

Guozhanxin, 2021-01-22:
已经合并了，关闭此issue

closed 2021-01-22 by Guozhanxin