RT-Thread/rt-thread #4119: How to use a 24-bit parallel LCD with  rt-thread ?

melkael, 2020-12-06:
Hello,

I'm a complete beginner with rt-thread. I would like to drive a 24-bit parallel RGB display for displaying a GUI using the rt-thread engine.
I'm using a F1C100S (Lichee pi Nano). However I'm a bit lost in the documentation, I believe my application is related to RT_Device_Class_Graphic but I can't figure out how to do things. Could you please provide guidance or an example ?

Thanks

BernardXiong, 2020-12-06:
You should write graphic driver with F1C100s datasheet, mostly of them are:

* TCON, DE setting;
* graphic driver with ops implemetation such as stm32f429 rgb lcd driver etc.

BernardXiong, 2020-12-06:
Another, this game boy uses F1C100s too:

[image]

Welcome to hack it, but it's lcd is a MCU lcd not the RGB lcd.

closed 2021-06-13 by mysterywolf