RT-Thread/rt-thread #4139: K20更改用户自己的交叉编译路径不起作用，注释以下语句可以起作用

chunyexixiaoyu, 2020-12-08:
`#if os.getenv('RTT_CC'):
#    CROSS_TOOL = os.getenv('RTT_CC')

if  CROSS_TOOL == 'gcc':
    PLATFORM    = 'gcc'
    EXEC_PATH   = r'D:/xpack-riscv-none-embed-gcc-8.3.0-1.2-win32-x64/xpack-riscv-none-embed-gcc-8.3.0-1.2/bin'
else:
    print('Please make sure your toolchains is GNU GCC!')
    exit(0)

#if os.getenv('RTT_EXEC_PATH'):
#    EXEC_PATH = os.getenv('RTT_EXEC_PATH')`

BernardXiong, 2020-12-09:
@balanceTWK Please take a look, thanks.

balanceTWK, 2020-12-09:
这个是临时解决方案：

需要在 env 终端里输入 `set RTT_EXEC_PATH=D:/xpack-riscv-none-embed-gcc-8.3.0-1.2-win32-x64/xpack-riscv-none-embed-gcc-8.3.0-1.2/bin`

balanceTWK, 2020-12-09:
好像不止一个 bsp 的  rtconfig.py 是这样的。晚上修复！

chunyexixiaoyu, 2020-12-09:
> 这个是临时解决方案：
> 
> 需要在 env 终端里输入 `set RTT_EXEC_PATH=D:/xpack-riscv-none-embed-gcc-8.3.0-1.2-win32-x64/xpack-riscv-none-embed-gcc-8.3.0-1.2/bin`

临时解决方案比较麻烦，readme里面也有临时解决方案

BernardXiong, 2020-12-09:
我粗略的感觉似乎没问题。这个(RT_)EXEC_PATH逻辑是这样的：

1. 如果环境变量里有`RTT_EXEC_PATH`，则优先使用环境变量的设置；
2. 如果环境变量中没有`RTT_EXEC_PATH`，则使用rtconfig.py中设置的EXEC_PATH；

类似的变量都是这样，（变量 =(优先使用)=> 环境变量）
* RTT_ROOT => RTT_ROOT
* CROSS_TOOL => RTT_CC
* EXEC_PATH => RTT_EXEC_PATH
* CC_PREFIX => RTT_CC_PREFIX <rt-smart开始使用>

balanceTWK, 2020-12-09:
> 我粗略的感觉似乎没问题。这个(RT_)EXEC_PATH逻辑是这样的：
> 
> 1. 如果环境变量里有`RTT_EXEC_PATH`，则优先使用环境变量的设置；
> 2. 如果环境变量中没有`RTT_EXEC_PATH`，则使用rtconfig.py中设置的EXEC_PATH；
> 
> 类似的变量都是这样，（变量 =(优先使用)=> 环境变量）
> 
> * RTT_ROOT => RTT_ROOT
> * CROSS_TOOL => RTT_CC
> * EXEC_PATH => RTT_EXEC_PATH
> * CC_PREFIX => RTT_CC_PREFIX <rt-smart开始使用>

@BernardXiong  现在 env 里默认会有 RTT_EXEC_PATH 环境变量。我现在 公司 家里 的电脑 env 都是这样的。

BernardXiong, 2020-12-09:
> @BernardXiong 现在 env 里默认会有 RTT_EXEC_PATH 环境变量。我现在 公司 家里 的电脑 env 都是这样的。

是的，env默认就会配置RTT_EXEC_PATH环境变量。所以如果需要一个独立的，针对risc-v的env版本，应该是有一套配置，把相关的配置更改为默认是RISC-V的。默认是一个*Init.cmd文件

balanceTWK, 2020-12-09:
CmdInit.cmd 里好像不好做判断是不是 risc-v 相关 bsp ，改当前 BSP 里的 rtconfig.py 我觉得可能会好改点。

balanceTWK, 2020-12-10:
我们内部做了讨论，讨论结果如下：

**修改 RISC-V 的 BSP 的 rtconfig.py. (也就是将 `EXEC_PATH = os.getenv('RTT_EXEC_PATH')` 相关的注释掉 )**

直接修改 env 工具，让 RTT_EXEC_PATH 的环境变量根据 不同的芯片架构 自动选择合适的工具链，这个其实是不合适的。因为要做到这一点，还需要给 env 添加 risc-v 工具链。这样会使得 env 体积变大。所以还是提倡开发者们再稍稍自行折腾下，自己下载合适的工具链，自己改工具链路径。

chunyexixiaoyu, 2020-12-10:
> 我们内部做了讨论，讨论结果如下：
> 
> **修改 RISC-V 的 BSP 的 rtconfig.py. (也就是将 `EXEC_PATH = os.getenv('RTT_EXEC_PATH')` 相关的注释掉 )**
> 
> 直接修改 env 工具，让 RTT_EXEC_PATH 的环境变量根据 不同的芯片架构 自动选择合适的工具链，这个其实是不合适的。因为要做到这一点，还需要给 env 添加 risc-v 工具链。这样会使得 env 体积变大。所以还是提倡开发者们再稍稍自行折腾下，自己下载合适的工具链，自己改工具链路径。

还有就是K210这个调试debug 官方出的IDE支持么？

balanceTWK, 2020-12-10:
> > 我们内部做了讨论，讨论结果如下：
> > **修改 RISC-V 的 BSP 的 rtconfig.py. (也就是将 `EXEC_PATH = os.getenv('RTT_EXEC_PATH')` 相关的注释掉 )**
> > 直接修改 env 工具，让 RTT_EXEC_PATH 的环境变量根据 不同的芯片架构 自动选择合适的工具链，这个其实是不合适的。因为要做到这一点，还需要给 env 添加 risc-v 工具链。这样会使得 env 体积变大。所以还是提倡开发者们再稍稍自行折腾下，自己下载合适的工具链，自己改工具链路径。
> 
> 还有就是K210这个调试debug 官方出的IDE支持么？

目前 rt-thread studio 已经有 riscv 的工具链，相应的 bsp 目前只有 gd32vf103 ， K210 的 bsp 还没有添加进去。但是未来肯定会有的。@chunyexixiaoyu

Guozhanxin, 2020-12-10:
[image]
其实准对这种非 ARM 架构芯片，env 自动工具链不能用的情况，不止 risc-v，其他的一些架构也有一些的。他们一般的处理方式都是通过设定 RTT_EXEC_PATH 环境变量的方式解决的。risc-v 的问题，也可以参考这些方式。

balanceTWK, 2020-12-10:
> [image]
> 其实准对这种非 ARM 架构芯片，env 自动工具链不能用的情况，不止 risc-v，其他的一些架构也有一些的。他们一般的处理方式都是通过设定 RTT_EXEC_PATH 环境变量的方式解决的。risc-v 的问题，也可以参考这些方式。

是的

closed 2020-12-10 by balanceTWK